coolsnowwolf/lede #7211: 请教各位如何把个性化的config文件编译到img里

gztingting, 2021-06-29:
记得以前谁教过把现在软路由上设置好的/etc/config/下的配置文件拷贝到拉取后的编译源文件哪个文件夹下，刷机时用squashfs文件系统，未来不管怎么整，只要firstboot一下，都会回到编译时设置好的一些配置。

我昨天试了一下，仓库目录下放一个DIY文件夹，下面放一个banner文件和一个config文件夹，下面有network、aria2、samba文件等，然后在云编译ssh步骤前的加载个性化配置里面加了一句 cp -rf ./DIY/* openwrt/package/base-files/files/etc/,结果编译出错，日志显示/etc/config下已经存在某某conf文件。这条路子走不通。

我在想，要不先把配置文件拷贝到/etc/tempconf临时文件夹，然后在package/lean/default-settings/files/zzz-default-settings最后加一句cp -rf  /etc/tempcon/* /etc/config/,不确定这个文件是不是写盘刷img文件后一次性运行的文件。

各位大神，你们是怎么做一些个性化设置的？

starxu1971, 2021-06-29:
在源码根目录下建立files文件夹把/etc/config/直接复制进去就好了编译时自己会调用的。就是这样的目录结构 files/etc/config/

gztingting, 2021-06-29:
> 在源码根目录下建立files文件夹把/etc/config/直接复制进去就好了编译时自己会调用的。就是这样的目录结构 files/etc/config/

原来是这样，云编译的yml文件还需要做任何调整吗？

starxu1971, 2021-06-29:
> > 在源码根目录下建立files文件夹把/etc/config/直接复制进去就好了编译时自己会调用的。就是这样的目录结构 files/etc/config/
> 
> 原来是这样，云编译的yml文件还需要做任何调整吗？

脚本不需要调整只要有这个目录

starxu1971, 2021-06-29:
https://github.com/281677160/AutoBuild-OpenWrt
你参考这个库，机型文件夹里的files文件夹就是这个作用

gztingting, 2021-06-29:
> https://github.com/281677160/AutoBuild-OpenWrt
> 你参考这个库，build文件夹里的files文件夹就是这个作用

你推荐的这个库，我有注意到，以前也用过，确实可以，只是他最近重建新库，还没搞明白他的用法。

目前我fork的是P3TERX/Actions-OpenWrt云编译，请教一下是直接在新fork的仓库下建立files文件夹，下面放自己想要的conf文件即可，对吗？

starxu1971, 2021-06-29:
> > https://github.com/281677160/AutoBuild-OpenWrt
> > 你参考这个库，build文件夹里的files文件夹就是这个作用
> 
> 你推荐的这个库，我有注意到，以前也用过，确实可以，只是他最近重建新库，还没搞明白他的用法。
> 
> 目前我fork的是P3TERX/Actions-OpenWrt云编译，请教一下是直接在新fork的仓库下建立files文件夹，下面放自己想要的conf文件即可，对吗？

是的

gztingting, 2021-06-29:
> > > https://github.com/281677160/AutoBuild-OpenWrt
> > > 你参考这个库，build文件夹里的files文件夹就是这个作用
> > 
> > 
> > 你推荐的这个库，我有注意到，以前也用过，确实可以，只是他最近重建新库，还没搞明白他的用法。
> > 目前我fork的是P3TERX/Actions-OpenWrt云编译，请教一下是直接在新fork的仓库下建立files文件夹，下面放自己想要的conf文件即可，对吗？
> 
> 是的，但要files/etc/config/这个目录结构

感激不尽

closed 2021-06-29 by gztingting

openwrt-try, 2021-07-02:
各位请教个问题，云编译时lean大佬源码里没有的luci-app怎么集成进去呀，比如adguardhome之类的