coolsnowwolf/lede #7213: docker not found

y0IkE, 2021-06-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
无法使用docker容器

### (1) 具体问题
A:
   root@OpenWrt:/etc# ./docker-web
  ./docker-web: line 3: docker: not found

B：
    root@OpenWrt:/etc# ln -s /etc/docker /usr/bin
    root@OpenWrt:/usr/bin# docker info
    -ash: docker: Permission denied

### (2) 路由器型号和固件版本

A：OpenWrt R21.6.22 / LuCI Master (git-21.175.10795-3d7bad0)

EnnawYang, 2021-06-29:
> root@OpenWrt:/etc# ln -s /etc/docker /usr/bin
> 
这是啥迷幻操作？

y0IkE, 2021-06-29:
> > root@OpenWrt:/etc# ln -s /etc/docker /usr/bin
> 
> 这是啥迷幻操作？

软链到bin，不然就出现A的那种情况

Saxon-Sun, 2021-06-29:
你把目录软链过去没用啊 把docker勾选就好了 luci-app-docker默认只选了dockerd  docker服务端 docker命令行工具

lisaac的解答：
luci-lib-docker: luci 调用 docker api 库，使用 dockerman 必须选
luci-app-dockerman: dockerman luci 面板，用于管理docker
luci-app-docker: lean写的luci界面
docker: docker命令行工具
dockerd: docker服务端

guaguagaga, 2021-06-29:
系统-软件包里装一下就好了

closed 2021-06-30 by coolsnowwolf