coolsnowwolf/lede #7215: 编译失败，提交报告

miansan, 2021-06-29:
编译环境Ubuntu 20，编译失败了，前几天都没问题，今天git pull然后make defconfig，编译失败，提交给lean大排错。

patching file drivers/net/wireless/ath/ath11k/peer.h
Hunk #1 succeeded at 9 (offset -11 lines).
Hunk #2 FAILED at 28.
1 out of 2 hunks FAILED -- saving rejects to file drivers/net/wireless/ath/ath11k/peer.h.rej
patching file drivers/net/wireless/ath/ath11k/pci.c
Hunk #1 succeeded at 530 (offset -219 lines).
Hunk #2 succeeded at 750 (offset -229 lines).
patching file drivers/net/wireless/ath/ath11k/debugfs.c
Hunk #1 succeeded at 643 (offset -279 lines).
patching file drivers/net/wireless/ath/ath11k/dp_tx.c
patching file drivers/net/wireless/ath/ath11k/dp_tx.h
patching file local-symbols
Hunk #1 succeeded at 153 (offset 20 lines).
Patch failed!  Please fix ./patches/nss/199-003-ath11k-add-nss-support.patch!
make[3]: *** [Makefile:579: /home/miansan/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.prepared_80fbf5680db44937324a308c4c44e5c9_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/miansan/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#1.07#0.38#1.39
    ERROR: package/kernel/mac80211 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/miansan/lede'
make[1]: *** [package/Makefile:110: /home/miansan/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/miansan/lede'
make: *** [/home/miansan/lede/include/toplevel.mk:230：world] 错误 2

miansan, 2021-06-29:
报告一下，没有make clean。

coolsnowwolf, 2021-06-29:
git pull

closed 2021-06-29 by coolsnowwolf