coolsnowwolf/lede #7218: 如何编译出小于4M的bin文件

jiatongwu, 2021-06-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：  我编译出来的bin文件9.3M  网上下载的不到4M  
       如何设置  menuconfig ?

wu@wu-VirtualBox:~/lede/bin/targets/ath79/tiny$ du -sh ./*
4.0K	./config.buildinfo
4.0K	./feeds.buildinfo
9.3M	./openwrt-ath79-tiny-tplink_tl-wr841-v7-initramfs-kernel.bin

macos@macosdeMacBook-Pro Downloads % du -sh lede-ar71xx-generic-tl-wr841-v7-squashfs-sysupgrade.bin 
4.0M    lede-ar71xx-generic-tl-wr841-v7-squashfs-sysupgrade.bin
macos@macosdeMacBook-Pro Downloads % 

### (2) 路由器型号和固件版本

A：tp 841n v7

### (3) 详细日志

A：

EnnawYang, 2021-06-29:
垃圾桶可能是最好的归宿

coolsnowwolf, 2021-06-29:
4MB 是不能刷入的，最大不能超过 3.25MB，因为还有 uboot，art，overlay 之类，否则启动都运行不起来

kisaragychihaya, 2021-06-29:
丢了吧，openwrt官方都不建议这设备了

xiaobaishu1, 2021-06-29:
哈哈哈 大雕都要把8M的给丢了  你现在好家伙出来个4M的设备

Cathgao, 2021-06-29:
这是在鉴宝还是在玩路由

Neverends4, 2021-06-30:
17.x可以支持4M的设备，你的841没记错就在其中。其实很简单，去掉Luci界面和一切不必要的插件，ssh+uci控制台调整一切即可。
当然自己整个大点的闪存焊上去。这个应该是NOR的，编程器也不贵，针脚也少~~~~

closed 2021-07-05 by jiatongwu