coolsnowwolf/lede #7219: 运行一段时间后，带宽监控无法正常监控流量

PyXiaoxin, 2021-06-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：nlbwmon 这个插件重启下就恢复了，但是过段时间又这样了。

### (2) 路由器型号和固件版本

A：
P&W R619AC 128M
OpenWrt R21.6.22 / LuCI Master (git-21.175.10795-3d7bad0)

### (3) 详细日志

A：报错日志如下
Tue Jun 29 00:59:42 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Out of memory
Tue Jun 29 00:59:42 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: No buffer space available
Tue Jun 29 01:00:00 2021 cron.err crond[29200]: USER root pid 3484 cmd echo "" > /tmp/vssr.log
Tue Jun 29 01:01:00 2021 cron.err crond[29200]: USER root pid 4258 cmd sh /usr/share/jd-dailybonus/newapp.sh -w
Tue Jun 29 01:54:48 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 01:54:49 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 01:55:19 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 01:55:19 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 02:07:20 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 02:07:20 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 02:08:20 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 02:08:20 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 02:10:20 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 02:10:20 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 02:12:20 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 02:12:20 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 02:14:20 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 02:14:20 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 02:15:50 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 02:15:50 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 02:20:53 2021 daemon.notice netifd: wan (11972): udhcpc: sending renew to 192.168.222.91
Tue Jun 29 02:20:58 2021 daemon.notice netifd: wan (11972): udhcpc: lease of 115.45.56.93 obtained, lease time 28800
Tue Jun 29 02:28:51 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 02:28:51 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 02:29:21 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 02:29:21 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 04:00:00 2021 cron.err crond[29200]: USER root pid 12202 cmd /etc/coremark.sh
Tue Jun 29 04:00:19 2021 daemon.err netdata[6571]: heartbeat missed 1500032 monotonic microseconds
Tue Jun 29 05:01:09 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Out of memory
Tue Jun 29 05:01:09 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: No buffer space available
Tue Jun 29 05:38:44 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 05:38:44 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 05:41:15 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 05:41:15 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 05:57:47 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 05:57:47 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 06:01:17 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 06:01:17 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 06:14:19 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 06:14:19 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 06:18:50 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 06:18:50 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 06:20:58 2021 daemon.notice netifd: wan (11972): udhcpc: sending renew to 192.168.222.91
Tue Jun 29 06:20:59 2021 daemon.notice netifd: wan (11972): udhcpc: lease of 115.45.56.93 obtained, lease time 28800
Tue Jun 29 06:29:51 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 06:29:51 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 07:00:25 2021 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED ce:c5:fa:9a:74:8a
Tue Jun 29 07:05:54 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Out of memory
Tue Jun 29 07:05:54 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: No buffer space available
Tue Jun 29 08:27:05 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 08:27:05 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 08:38:37 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 08:38:37 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 08:51:09 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 08:51:09 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 08:59:10 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 08:59:10 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 09:08:41 2021 daemon.err nlbwmon[10002]: Netlink receive failure: Object busy
Tue Jun 29 09:08:41 2021 daemon.err nlbwmon[10002]: Unable to dump conntrack: I/O error
Tue Jun 29 15:00:00 2021 cron.err crond[29200]: USER root pid 5035 cmd /usr/bin/serverchan/serverchan send &
Tue Jun 29 15:00:01 2021 cron.err crond[5204]: crond (busybox 1.33.1) started, log level 5

coolsnowwolf, 2021-06-29:
Netlink receive failure: Out of memory 内存不足

PyXiaoxin, 2021-06-30:
@coolsnowwolf 但是内存一直都没满过，剩余70%

Neverends4, 2021-06-30:
nlbwmon经常报内存不足，试试设置
/etc/config/nlbwmon: option netlink_buffer_size 10485760
然后重启服务

PyXiaoxin, 2021-06-30:
@Neverends4 好滴，感谢大佬~我试试

minlang112, 2021-06-30:
> nlbwmon经常报内存不足，试试设置
> /etc/config/nlbwmon: option netlink_buffer_size 10485760
> 然后重启服务

确实，看日志就是报存储空间不足

realconnor, 2021-07-01:
个人经验，改缓存大小效果不大，这与网络活动强度有关，比如我在后面加个零还是会报缓存溢出，尤其是挂PT的情况下，因为默认的提交间隔是24小时。
比较合适的方法是分析下日志，看看报溢出的最小时间间隔是多少，然后把配置里的“提交间隔”改成小于该间隔的一个整数，我是设置成1小时，再没出过问题。从你的日志来看，设置成2小时应该就没事了，搭配更大的缓存可以增大这个间隔，以减小频繁写入数据导致的闪存损耗（非常微小）。

closed 2024-11-19 by coolsnowwolf