coolsnowwolf/lede #7230: luci-app-easymesh 打包失败

WYC-2020, 2021-06-30:
编译完毕，在进行最后打包的时候出现以下错误：
Collected errors:
 * check_conflicts_for: The following packages conflict with wpad-openssl:
 * check_conflicts_for: 	wpad * 
 * opkg_install_cmd: Cannot install package wpad-openssl.
 * check_conflicts_for: The following packages conflict with wpad-openssl:
 * check_conflicts_for: 	wpad * 
 * opkg_install_cmd: Cannot install package luci-app-easymesh.

coolsnowwolf, 2021-06-30:
不能同时安装2个 wpad

WYC-2020, 2021-06-30:
@coolsnowwolf 是luci-app-easymesh这个程序依赖的wpad-openssl 和另外某一个程序的依赖有冲突吧，目测应该是hostapd这个库有冲突，暂时不知咋解决，临时方案吧luci-app-easymesh依赖wpad-openssl 去掉编译通过

closed 2021-06-30 by WYC-2020

swtatoyang, 2021-07-17:
> @coolsnowwolf 是luci-app-easymesh这个程序依赖的wpad-openssl 和另外某一个程序的依赖有冲突吧，目测应该是hostapd这个库有冲突，暂时不知咋解决，临时方案吧luci-app-easymesh依赖wpad-openssl 去掉编译通过

兄，要怎么去掉wpad-openssl编译呢？

Houiin, 2021-09-18:
> > @coolsnowwolf 是luci-app-easymesh这个程序依赖的wpad-openssl 和另外某一个程序的依赖有冲突吧，目测应该是hostapd这个库有冲突，暂时不知咋解决，临时方案吧luci-app-easymesh依赖wpad-openssl 去掉编译通过
> 
> 兄，要怎么去掉wpad-openssl编译呢？

同问...

Houiin, 2021-09-18:
> > @coolsnowwolf 是luci-app-easymesh这个程序依赖的wpad-openssl 和另外某一个程序的依赖有冲突吧，目测应该是hostapd这个库有冲突，暂时不知咋解决，临时方案吧luci-app-easymesh依赖wpad-openssl 去掉编译通过
> 
> 兄，要怎么去掉wpad-openssl编译呢？

刚刚尝试了一下，修改`lede/.config` 文件，注释掉`CONFIG PACKAGE wpad` 那一行就可以编译成功