coolsnowwolf/lede #7236: k2p重启后丢失配置, 并且无法生成备份

dengxin6, 2021-06-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：1.路由器重启后丢失配置; 2.无法生成备份; 3.无法通过sysupgrade直接更新固件

### (2) 路由器型号和固件版本

A：k2p 5.4.128

### (3) 详细日志

A：
Sat Jun 26 02:48:40 2021 kern.notice kernel: [    0.000000] Linux version 5.4.128 (runner@fv-az297-33) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3356-b8abbb685)) #0 SMP Fri Jun 25 18:48:33 2021
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] MIPS: machine is Phicomm K2P
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sat Jun 26 02:48:40 2021 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Jun 26 02:48:40 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sat Jun 26 02:48:40 2021 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Zone ranges:
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000]   HighMem  empty
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Early memory node ranges
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jun 26 02:48:40 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Sat Jun 26 02:48:40 2021 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Sat Jun 26 02:48:40 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sat Jun 26 02:48:40 2021 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
Sat Jun 26 02:48:40 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
Sat Jun 26 02:48:40 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Sat Jun 26 02:48:40 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=0000ce00
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=0000ce00
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] Memory: 108808K/131072K available (6180K kernel code, 257K rwdata, 1296K rodata, 12220K init, 238K bss, 22264K reserved, 0K cma-reserved, 0K highmem)
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] NR_IRQS: 256
Sat Jun 26 02:48:40 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] CPU Clock: 1000MHz
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 3822520893 ns
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.000008] sched_clock: 32 bits at 500MHz, resolution 2ns, wraps every 4294967295ns
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.015490] Calibrating delay loop... 666.41 BogoMIPS (lpj=3332096)
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.087809] pid_max: default: 32768 minimum: 301
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.097129] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.111529] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.129453] rcu: Hierarchical SRCU implementation.
Sat Jun 26 02:48:40 2021 kern.info kernel: [    0.139529] smp: Bringing up secondary CPUs ...
Sat Jun 26 02:48:40 2021 kern.warn kernel: [    7.074699] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Jun 26 02:48:40 2021 kern.warn kernel: [    7.074709] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sat Jun 26 02:48:40 2021 kern.warn kernel: [    7.074720] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[... 373 lines cut ...]
Sat Jun 26 02:49:19 2021 daemon.notice netifd: Interface 'wan6' has link connectivity
Sat Jun 26 02:49:19 2021 daemon.notice netifd: Interface 'wan6' is setting up now
Sat Jun 26 02:49:19 2021 daemon.notice procd: /etc/rc.d/S60flowoffload: udhcpc: started, v1.33.1
Sat Jun 26 02:49:19 2021 daemon.err odhcp6c[3139]: Failed to send RS (Address not available)
Sat Jun 26 02:49:19 2021 daemon.notice netifd: wan (3154): udhcpc: started, v1.33.1
Sat Jun 26 02:49:19 2021 daemon.notice procd: /etc/rc.d/S60flowoffload: udhcpc: sending discover
Sat Jun 26 02:49:19 2021 daemon.notice netifd: wan (3154): udhcpc: sending discover
Sat Jun 26 02:49:19 2021 authpriv.warn dropbear[3192]: Failed listening on '22': Error listening: Address not available
Sat Jun 26 02:49:19 2021 authpriv.info dropbear[3192]: Not backgrounding
Sat Jun 26 02:49:20 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jun 26 02:49:20 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jun 26 02:49:20 2021 daemon.err odhcp6c[3139]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sat Jun 26 02:49:20 2021 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sat Jun 26 02:49:21 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan (br-lan)
Sat Jun 26 02:49:22 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Sat Jun 26 02:49:22 2021 daemon.notice netifd: wan (3154): udhcpc: sending discover
Sat Jun 26 02:49:23 2021 daemon.notice procd: /etc/rc.d/S60flowoffload: udhcpc: no lease, failing
Sat Jun 26 02:49:23 2021 daemon.info dnsmasq[3933]: Connected to system UBus
Sat Jun 26 02:49:23 2021 daemon.notice procd: /etc/rc.d/S60flowoffload: DNSMASQ change
Sat Jun 26 02:49:25 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jun 26 02:49:25 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jun 26 02:49:25 2021 daemon.notice netifd: wan (3154): udhcpc: sending discover
Sat Jun 26 02:49:28 2021 daemon.info dnsmasq[4454]: Connected to system UBus
Sat Jun 26 02:49:29 2021 daemon.info procd: - init complete -
Sat Jun 26 02:49:29 2021 daemon.info urandom_seed[4665]: Seed saved (/etc/urandom.seed)

jlzsss, 2021-06-30:
x86版也有同样给的问题。

ltactor, 2021-07-01:
k2p 32m 重启变红灯，应该是有个模式设置问题。

chazliu, 2021-09-17:
这两天编译的也是这样，斐讯和公版都试了，breed也是最新的，无论先ramfs再sysupgrade还是直接sysupgrade都是闪一下红灯然后长蓝，就是启动不了。。。

ChuJian2, 2022-01-21:
现在还能正常编译吗？我按照上面提到的方法来编译，编译出来好像无法识别闪存。