coolsnowwolf/lede #7244: 最新编译的固件不能上网，截取了一段系统日志

qqsir-dev, 2021-07-01:
反馈bug/问题模板，提建议请删除
我刷新固件没有保留设置，但是还是不能上网
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
Thu Jul  1 19:11:03 2021 daemon.notice procd: /etc/rc.d/S90turboacc: net.ipv4.tcp_congestion_control = bbr
Thu Jul  1 19:11:03 2021 daemon.notice procd: /etc/rc.d/S90turboacc: killall: dnscache: no process killed
Thu Jul  1 19:11:03 2021 daemon.err collectd[3775]: plugin_load: plugin "rrdtool" successfully loaded.
Thu Jul  1 19:11:03 2021 daemon.err collectd[3775]: rrdtool plugin: RRASingle = true: creating only AVERAGE RRAs
Thu Jul  1 19:11:03 2021 daemon.err collectd[3775]: plugin_load: plugin "interface" successfully loaded.
Thu Jul  1 19:11:03 2021 daemon.err collectd[3775]: Initialization complete, entering read-loop.
Thu Jul  1 19:11:03 2021 daemon.notice procd: /etc/rc.d/S90turboacc: sh: you need to specify whom to kill
Thu Jul  1 19:11:03 2021 daemon.notice procd: /etc/rc.d/S90turboacc: Stop DNS Caching
Thu Jul  1 19:11:03 2021 authpriv.warn dropbear[4272]: Failed listening on '22': Error listening: Address not available
Thu Jul  1 19:11:03 2021 authpriv.info dropbear[4272]: Not backgrounding
Thu Jul  1 19:11:04 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Thu Jul  1 19:11:04 2021 daemon.notice procd: /etc/rc.d/S90turboacc: PDNSD: Start DNS Caching
Thu Jul  1 19:11:04 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jul  1 19:11:04 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jul  1 19:11:04 2021 daemon.notice procd: /etc/rc.d/S90turboacc: DNSMASQ change
Thu Jul  1 19:11:04 2021 daemon.info dnsmasq[4635]: Connected to system UBus
Thu Jul  1 19:11:04 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jul  1 19:11:04 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jul  1 19:11:04 2021 daemon.info dnsmasq[4930]: Connected to system UBus
Thu Jul  1 19:11:05 2021 daemon.notice netifd: Network device 'eth0' link is down
Thu Jul  1 19:11:05 2021 kern.info kernel: [   15.383774] br-lan: port 1(eth0) entered disabled state
Thu Jul  1 19:11:05 2021 daemon.notice netifd: Network device 'eth0' link is up
Thu Jul  1 19:11:05 2021 kern.info kernel: [   15.402591] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
Thu Jul  1 19:11:05 2021 kern.info kernel: [   15.403436] br-lan: port 1(eth0) entered blocking state
Thu Jul  1 19:11:05 2021 kern.info kernel: [   15.403584] br-lan: port 1(eth0) entered forwarding state
Thu Jul  1 19:11:06 2021 user.info mwan3rtmon[3745]: Detect rtchange event.
Thu Jul  1 19:11:07 2021 user.info wsdd2[5392]: starting.
Thu Jul  1 19:11:07 2021 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Thu Jul  1 19:11:07 2021 daemon.info procd: - init complete -
Thu Jul  1 19:11:25 2021 kern.err kernel: [   35.724315] squashfs: Unknown parameter 'discard'
Thu Jul  1 19:11:25 2021 kern.info kernel: [   35.724703] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
Thu Jul  1 19:11:25 2021 kern.err kernel: [   35.724820] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
Thu Jul  1 19:11:25 2021 kern.info kernel: [   35.725075] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x80a345bd)
Thu Jul  1 19:11:25 2021 kern.err kernel: [   35.725207] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
Thu Jul  1 19:11:25 2021 kern.err kernel: [   35.726132] exFAT-fs (sda128): invalid boot record signature
Thu Jul  1 19:11:25 2021 kern.err kernel: [   35.726220] exFAT-fs (sda128): failed to read boot sector
Thu Jul  1 19:11:25 2021 kern.err kernel: [   35.726302] exFAT-fs (sda128): failed to recognize exfat type
Thu Jul  1 19:11:52 2021 daemon.err rmmod: module is not loaded
Thu Jul  1 19:11:52 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jul  1 19:11:52 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jul  1 19:11:52 2021 daemon.info dnsmasq[7601]: Connected to system UBus
Thu Jul  1 19:12:10 2021 daemon.err rmmod: module is not loaded
Thu Jul  1 19:12:10 2021 daemon.info dnsmasq[8150]: Connected to system UBus
Thu Jul  1 19:12:50 2021 daemon.info dnsmasq[8862]: Connected to system UBus
Thu Jul  1 19:12:55 2021 daemon.err uhttpd[3420]: Failed to send request: Operation not permitted
Thu Jul  1 19:12:55 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jul  1 19:12:55 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jul  1 19:12:55 2021 daemon.info dnsmasq[9055]: Connected to system UBus
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

WangQi107, 2021-07-03:
进pb-boot刷进去就正常了，我自己试过了，之前也是直接op里升级后没法上网

WangQi107, 2021-07-03:
不过问题是所有的东西要重新配置一遍

qqsir-dev, 2021-07-04:
PBboot是什么?我是个老白，能教教我吗，谢谢

qqsir-dev, 2021-07-04:
我好下像找到原因了，旧版的Turbo ACC里的DNS Caching设置的是内网 AD Guard Home的IP，新版这样就不行，而且不改设置不打开也不行，一定要用公网DNS才可以，这是问题还是变化? 求解？

WangQi107, 2021-07-04:
pb-boot和breed功能一样刷固件的，我的问题是WAN和LAN接口都无法连接

closed 2021-07-05 by qqsir-dev