coolsnowwolf/lede #7245: 部分软件包运行后提示“extra_command：not found”报错

dansir118, 2021-07-01:
## 1.关于你要提交的问题
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue
## 2. 详细叙述
### (1) 具体问题
目前发现dnsforwarder、vpnbypass、zram软件包运行后提示“extra_command：not found”报错。
猜测是这条commit导致：https://github.com/coolsnowwolf/lede/commit/4a0fbb84adaf0bf553cfcf81a8fe178d9ffdb601
### (2) 路由器型号和固件版本
ASUS RT-ACRH17/R21.6.22-Linux5.4.128-git_a9b30b2
### (3) 详细日志

```
daemon.err uhttpd[3712]: sh: /etc/init.d/dnsforwarder: line 5: extra_command: not found
daemon.err uhttpd[3712]: sh: /etc/init.d/dnsforwarder: line 6: extra_command: not found
daemon.err uhttpd[3712]: sh: /etc/init.d/dnsforwarder: line 7: extra_command: not found
daemon.err uhttpd[3712]: sh: /etc/init.d/vpnbypass: line 12: extra_command: not found
daemon.err uhttpd[3712]: sh: /etc/init.d/zram: line 5: extra_command: not found
daemon.err uhttpd[3712]: sh: /etc/init.d/zram: line 6: extra_command: not found
```

dansir118, 2021-07-01:
看到这条已合并的PR已经增加了“extra_command”方法，做了兼容：https://github.com/coolsnowwolf/lede/pull/6357
不知道为啥没生效，在部分软件包中还是提示"extra_command: not found"

coolsnowwolf, 2021-07-02:
不影响运行，确实介意的自己做个 PR 按照我的方法用老版本的 API，双向兼容的

dansir118, 2021-07-02:
> 不影响运行，确实介意的自己做个 PR 按照我的方法用老版本的 API，双向兼容的

照葫芦画瓢，已提交PR https://github.com/coolsnowwolf/lede/pull/7259

closed 2021-07-03 by dansir118