coolsnowwolf/lede #7246: luci-app-dnsfilter自带规则合并出错

zwillhill, 2021-07-01:
## 1.合并规则功能有bug

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：原版dnsfilter前三条自带规则（anti-AD、Adguard、Easylist）更新完成后约5w条，merge版同样的规则更新完后体积翻倍约14w条。
多出来的规则格式出错，国内网站正常，bypass出国插件无法运行无法连接服务器，op自带日志提示rules.conf首字符（$）出错。

正常规则格式
```
address=/0024aaaa.com/
address=/0024aadd.com/
address=/003store.com/
address=/006a039c957c142bb.com/
address=/0082tv.net/
```

不正常规则格式

```
&AdType=/
&Ads_DFP=/
&UrlAdParam=/
+advertorial./
+adverts//
-ad-300x250./
-ad-300x450./
-ad-300x600-/
greatergood.com/
greatis.com/
greatwuyi.com/
||zzznews.ru^
||zzzyk.com/js//
~05wang.com/
```

### (2) 路由器型号和固件版本

A：k3，0701云编译版固件

### (3) 详细日志

A：

zwillhill, 2021-07-01:
[image]

small-5, 2021-07-01:
你用我仓库里面的那个就ok了，adblock-plus

small-5, 2021-07-01:
你用我仓库里面的那个就ok了，adblock-plus。dnsfilter的原作者是我，某个我不好评论的人改个名字就好像变成他的项目了

closed 2021-07-01 by zwillhill