coolsnowwolf/lede #7250: AX6 sirq占用高的问题

ldy8182, 2021-07-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
没有搜到
## 2. 详细叙述
AX6 sirq占用高的问题
### (1) 具体问题

A：开启Turbo ACC 网络加速设置-软件流量分载-全锥形 NAT，sirq占用更高，关闭后会降低一些

### (2) 路由器型号和固件版本

A：Redmi AX6  OpenWrt R21.7.1.100 / LuCI Master (git-21.181.13804-f7b5f6d)

### (3) 详细日志

A：
[image]
[image]
[系统日志.txt](https://github.com/coolsnowwolf/lede/files/6752305/default.txt)
[内核日志.txt](https://github.com/coolsnowwolf/lede/files/6752306/default.txt)

ldy8182, 2021-07-02:
很可能是NFT-QoS这个软件的问题，我关掉后重启路由器就正常了

coolsnowwolf, 2021-07-02:
nss 不兼容 ntf

ldy8182, 2021-07-02:
感谢大雕，我把这个软件去掉~

closed 2021-07-02 by ldy8182

xsm1997, 2021-07-06:
@ldy8182 请问您sirq高的问题是怎么解决的？我没编译NFT-QoS这个软件，软转发时sirq也很高，有20%多，且软转发只有300M左右。

ldy8182, 2021-07-06:
qos类的都去掉，流量监控也去掉

xsm1997, 2021-07-06:
@ldy8182 您现在使用时sirq大概是多少呢？

ldy8182, 2021-07-06:
10%-15%