coolsnowwolf/lede #7253: 全新环境全新编译x86   qbittorrent报错

c939984606, 2021-07-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： 全新环境全新编译，qbittorrent报错

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：check_data_file_clashes: Package qbittorrent wants to install file /home/kcui/Openwrt/lede/build_dir/target-x86_64_musl/root-x86/usr/bin/qbittorrent-nox
But that file is already provided by package * qBittorrent-static
opkg_install_cmd: Cannot install package qbittorrent.
make[2]: *** [package/Makefile:70: package/install] Error 255

c939984606, 2021-07-02:
commit 01e4998f6d4e9604ab101a897acae7a52f20a991

package/lean/qBittorrent-static/Makefile

define Package/$(PKG_NAME)/install
	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(DL_DIR)/qbittorrent-nox $(1)/usr/bin
endef

closed 2021-07-02 by coolsnowwolf

sheno-Tyes, 2021-07-04:
解决了？

wangyang8284, 2021-07-04:
好像没解决，还是报错

openwrt-try, 2021-07-05:
> commit [01e4998](https://github.com/coolsnowwolf/lede/commit/01e4998f6d4e9604ab101a897acae7a52f20a991)
> 
> package/lean/qBittorrent-static/Makefile
> 
> define Package/$(PKG_NAME)/install
> $(INSTALL_DIR) $(1)/usr/bin
> $(INSTALL_BIN) $(DL_DIR)/qbittorrent-nox $(1)/usr/bin
> endef

内容跟这上面的一样，
还是报这个错误，不管选不选luci-app-qBittorrent，
求指点

sheno-Tyes, 2021-07-05:
需要在network->bittorrent,取消勾选qbittorrent或qbittorrent-static,选一个

wangyang8284, 2021-07-05:
> 需要在network->bittorrent,取消勾选qbittorrent或qbittorrent-static,选一个

感谢感谢！