coolsnowwolf/lede #7254: SoftEther VPN Server连接失败

minfengyk, 2021-07-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
    SoftEther VPN Server连接失败
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
    7月1日编译的固件，SoftEther管理器连接SoftEther VPN Server失败。
### (1) 具体问题

A：    7月1日编译的固件，DD刷机全新安装。路由器中SoftEther VPN成功开启，但设置SoftEther VPN时，管理器连接SoftEther VPN Server失败，不能进一步设置。

### (2) 路由器型号和固件版本

A：
主机型号 | Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz : 1 Core 2 Thread
CPU 信息 | 2400.000 MHz
固件版本 | OpenWrt R21.7.1.100 / LuCI Master (git-21.181.13804-f7b5f6d)
内核版本 | 5.10.46

### (3) 详细日志

A：Fri Jul  2 17:27:54 2021 user.notice firewall: Reloading firewall due to ifup of wan1 (pppoe-wan1)
Fri Jul  2 17:27:54 2021 user.notice softethervpn: Starting softether vpnserver service.
Fri Jul  2 17:27:54 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan1 (pppoe-wan1)
Fri Jul  2 17:27:54 2021 user.notice ddns-scripts[6026]: aliyun_ipv4: PID '6026' started at 2021-07-02 17:27
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn: Warning: Section @zone[1] (wan) cannot resolve device of network 'wan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn: Warning: Section @zone[1] (wan) cannot resolve device of network 'wan6'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:  * Clearing IPv4 filter table
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:  * Clearing IPv4 nat table
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:  * Clearing IPv4 mangle table
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:  * Clearing IPv4 raw table
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:  * Populating IPv4 filter table
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Rule 'Allow-DHCP-Renew'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Rule 'Allow-Ping'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Rule 'Allow-IGMP'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Rule 'Allow-IPSec-ESP'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Rule 'Allow-ISAKMP'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Rule 'openvpn'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Rule 'kms'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Aria2 RPC'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Aria2 DHT'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Secure Web Serve'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Secure Web Admin'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Web Serve'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Web Admin'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Forward 'lan' -> 'wan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Forward 'vpn' -> 'wan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Forward 'vpn' -> 'lan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Forward 'lan' -> 'vpn'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Zone 'lan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Zone 'wan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Zone 'vpn'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:  * Populating IPv4 nat table
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Aria2 RPC'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Aria2 DHT'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Secure Web Serve'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Secure Web Admin'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Web Serve'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Redirect 'Web Admin'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Zone 'lan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Zone 'wan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn: Warning: wan will enable FULLCONE-NAT
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Zone 'vpn'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn: Warning: vpn will enable FULLCONE-NAT
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:  * Populating IPv4 mangle table
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Zone 'lan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Zone 'wan'
Fri Jul  2 17:27:54 2021 daemon.notice procd: /etc/rc.d/S99softethervpn:    * Zone 'vpn'
[... 55 lines cut ...]
Fri Jul  2 17:27:55 2021 user.info mwan3rtmon[3664]: Detect rtchange event.
Fri Jul  2 17:28:19 2021 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Fri Jul  2 17:38:03 2021 user.warn ddns-scripts[6026]: aliyun_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Fri Jul  2 17:41:21 2021 cron.err crond[18374]: crond (busybox 1.33.1) started, log level 5
Fri Jul  2 17:41:21 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Jul  2 17:41:21 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Jul  2 17:41:21 2021 daemon.info dnsmasq[18523]: Connected to system UBus
Fri Jul  2 17:41:21 2021 cron.err crond[18676]: crond (busybox 1.33.1) started, log level 5
Fri Jul  2 17:41:27 2021 cron.err crond[18977]: crond (busybox 1.33.1) started, log level 5
Fri Jul  2 17:41:27 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Jul  2 17:41:27 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Jul  2 17:41:27 2021 daemon.info dnsmasq[19127]: Connected to system UBus
Fri Jul  2 17:41:27 2021 cron.err crond[19280]: crond (busybox 1.33.1) started, log level 5
Fri Jul  2 17:42:15 2021 cron.err crond[19591]: crond (busybox 1.33.1) started, log level 5
Fri Jul  2 17:42:15 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Jul  2 17:42:15 2021 daemon.info dnsmasq[19739]: Connected to system UBus
Fri Jul  2 17:42:15 2021 cron.err crond[19892]: crond (busybox 1.33.1) started, log level 5
Fri Jul  2 17:42:33 2021 cron.err crond[20518]: crond (busybox 1.33.1) started, log level 5
Fri Jul  2 17:42:33 2021 daemon.info dnsmasq[20665]: Connected to system UBus
Fri Jul  2 17:42:34 2021 daemon.info pdnsd[20846]: pdnsd-1.2.9b-par starting.
Fri Jul  2 17:42:34 2021 cron.err crond[21738]: crond (busybox 1.33.1) started, log level 5
Fri Jul  2 17:42:35 2021 daemon.info dnsmasq[21640]: Connected to system UBus
Fri Jul  2 17:43:32 2021 daemon.err uhttpd[3415]: netstat: showing only processes with your user ID
Fri Jul  2 17:48:05 2021 user.warn ddns-scripts[6026]: aliyun_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Fri Jul  2 17:58:06 2021 user.warn ddns-scripts[6026]: aliyun_ipv4: Updating IP at DDNS provider failed - starting retry 3/0

lexin8, 2021-07-02:
不纯在你说的问题

minfengyk, 2021-07-02:
找到原因了：默认连接端口443被占用，端口改为992或5555，就可以了。

closed 2021-08-15 by AmadeusGhost