coolsnowwolf/lede #7261: 希望能恢复K2P 32M的选项

soundghost, 2021-07-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：大神，您近期的版本删除了斐讯K2P 32M的固件，我只得又按照以前的方法修改dts的分区大小和mt7621.mk的固件尺寸。由于分区大小不一样，虽然最终能生成32M固件，但是无法网页升级，我用breed刷入initramfs固件后，再从网页升级squashfs固件后，发现固件还是“只读”，也就是没有恢复设置的选项。
看来我用老办法是不行的了。
因此只得上来提交issues，希望您能够恢复K2P的32M固件选项，毕竟硬改K2P的玩家也不少。

万分感谢您！！

Boos4721, 2021-07-02:
https://github.com/coolsnowwolf/lede/commit/f6381d20944098e0079eccc18737bd2ef2ec4d7b
https://github.com/coolsnowwolf/lede/commit/18d284b238f2f2b2f75e6492330858a25a21b865
pick一下

yinzhefuyun, 2021-07-03:
这里解释一下为什么lean他们要删K2P 32MB，都是被奸商恶心的，因为每个人用的32MB的SPI芯片不一致，导致开机引导的驱动也是不一致的（重点是奸商还把锅推给lean他们），举个例子歌华链官方用来两个SPI芯片MX25L25635E和MX25L25635F（就最后一位不一致）就导致无论dts配置哪个一个驱动都会导致另外一个要不然无法开机，要不然无法软重启，好在最后找到了高通的开源SPI驱动刚刚好这两个SPI是用同一个驱动才解决的，那你可以想一想每家每户的K2P用的SPI别说型号了很多连厂家都不一样，然后天天骂为什么今天我能启动，明天他能启动，这个SPI驱动非常底层无法在加载前做判断，再加上各各参加和型号一大堆，怎么做都是被骂，所以直接删了

soundghost, 2021-07-05:
谢谢楼上两位

ltactor, 2021-07-07:
我是自己改的32M， 试过上面的pick无法编译成功。请问给位老师有成功的吗？

yinzhefuyun, 2021-07-08:
> 谢谢楼上两位

看到上面那条了吗，不同的SPI就有兼容问题，这个就是pick后不能软重启的例子，每个人都是去淘宝随便买一个32MB的SPI，然后都麻烦AmadeusGhost做兼容，然后今天兼容你就不兼容他，明天他就发issues说我的出问题了，你现在明白了为什么不做K2P-32MB兼容了吧

ltactor, 2021-07-08:
> > 谢谢楼上两位
> 
> 看到上面那条了吗，不同的SPI就有兼容问题，这个就是pick后不能软重启的例子，每个人都是去淘宝随便买一个32MB的SPI，然后都麻烦AmadeusGhost做兼容，然后今天兼容你就不兼容他，明天他就发issues说我的出问题了，你现在明白了为什么不做K2P-32MB兼容了吧

之前的k2p 32M 可以启动。    一直也没有出现问题，之前应该是解决了兼容问题。   最近删除了32m ，才不可以的。
只是可惜k2p 32M   只能做旁路由了。

soundghost, 2021-07-10:
> 我是自己改的32M， 试过上面的pick无法编译成功。请问给位老师有成功的吗？
> @AmadeusGhost
> 麻烦老师，取消32M太可惜了。我是自己改的32M 主要是用来编译wireguard联回家里。
> 这次更新编译后无法重启（首次刷入可以启动，保存配置重启常亮红灯），上次出这问题时好像是更改&flash { broken-flash-reset;}; 解决的。 请问老师这次怎用修改？
> 我pick上面的提交，无法编译成功。
> 我觉得k2p的余热还没有发挥完，希望还有机会修复。感谢。
其实我也没有成功，在没有删除32M选项前，编译出来的32M固件确实可以启动，软重启也是完美的。
但是由于那个时候我没有开通ipv6，所以后来才想要重新编译一版继承了ipv6helper的固件，然后就不行了。
我按照文件的内容手动修改了固件大小、还有分区大小，编译出来的固件直接网页升级sys固件时提示不支持该固件，进breed先刷int固件（公版分区），重启后再上传sys固件，重启后还是不能出现恢复出厂设置的按钮（似乎刷入的sys固件无效），并且选择argon主题的时候css是错位的。重进breed选择斐讯分区刷入int固件，argon主题正常了，刷入sys固件重启，还是没有恢复出厂设置的按钮。
如果这样把文件修改为之前的参数都还是不能正常使用的话，这个K2P我就准备停留在之前的版本不考虑用它来上ipv6了（刷机前我做了编程器固件备份的）。

ltactor, 2021-07-13:
> 其实我也没有成功，在没有删除32M选项前，编译出来的32M固件确实可以启动，软重启也是完美的。
> 但是由于那个时候我没有开通ipv6，所以后来才想要重新编译一版继承了ipv6helper的固件，然后就不行了。 [...]

@AmadeusGhost 你好，我是自己改装的32M ，  之前使用未合并版本，可以软起动，现最新版本无法重启。我觉得肯能是&flash { broken-flash-reset;};  问题。  我尝试过好几次还是改不对，请教老师怎样修改才能使用32M。

shichunjian, 2021-07-16:
> > > 我是自己改的 32M，试过上面的选择无法成功。请问给位老师有成功的吗？
> > > @AmadeusGhost 麻烦
> > > 老师，取消 32M 太可惜了。我是自己改的 32M 主要是编译线卫联回这一次
> > > 更新后无法重启（第一次刷入可以启动，重启重启常亮红灯），上次出这个问题时就好像更改了&flash {break-flash-reset;}; 解决的。如老师这
> > > 能用修改吗？我选择了上面的提交，无法合成成功。
> > > 我觉得 k2p 的余热还完成了，希望还有机会修复。感谢。
> > 
> > 
> > 实际上我也没有成功，在删除 32M 选项前的 32M 选项前，没有重新启动，软重启完美的。但由于那个
> > 我没有开通 ipv6 的时候，所以才需要编译一版继承了 ipv6helper 的固件，然后就不行了。
> > 我按照文件的内容手动修改了固件大小，还有分区大小，编译出来的固件直接网页升级SYS固件时提示不支持该固件，进滋生先刷INT固件（公版分区） ，重启再上传系统触发，重启后不能出现恢复能力设置的按钮（自己刷入系统触发无效），并且选择氩气的主题时机是错的。 ，不能再这样下去了，刷入重启重启，
> > 如果还没有这样的参数还是这个系统设置的按钮。如果没有这样的参数就可以正常使用的话，K2P我准备在以后不考虑用它来上ipv6了（刷机前我这编程器伙伴的）。
> 
> @AmadeusGhost你好，我是自己安装的32M，之前使用未合并版本，可以软启动，最新版本重启。我尝试过好几次还是有些不对劲，请教老师怎么修改才能使用32M。

你把内核改为5.4就行了，5.10内核下有问题。

ltactor, 2021-07-19:
> 你把内核改为5.4就行了，5.10内核下有问题。

感谢回复， 请问怎样将内核改为5.4 。    新手有点茫然。。。。

soundghost, 2021-07-26:
> 你把内核改为5.4就行了，5.10内核下有问题。

今晚将内核改为了5.4还是不行，自从删除掉32M的K2P选项后，即使将dts和mt7621.mk文件做了相应的改动之后，生成的固件还是没有恢复出厂设置的选项。

我的刷机步骤是先breed刷入initramfs固件，然后在luci里升级squashfs固件，可是就算在这里升级了squashfs固件重启后，依然没有恢复出厂设置那个按钮。同时系统-软件包里的剩余空间也显示异常，显示剩余167MB

@ltactor  

这里有你需要的资料[https://www.bgegao.com/2020/11/1885.html](https://www.bgegao.com/2020/11/1885.html)

shichunjian, 2021-07-27:
> 今晚将内核改为了5.4还是不行，自从删除掉32M的K2P选项后，即使将dts和mt7621.mk文件做了相应的改动之后，生成的固件还是没有恢复出厂设置的选项。
>
> 我的刷机步骤是先breed刷入initramfs固件，然后在luci里升级squashfs固件，可是就算在这里升级了squashfs固件重启后，依然没有恢复出厂设置那个按钮。同时系统-软件包里的剩余空间也显示异常，显示剩余167MB [...]

sed -i 's/15744k/32128k/g' target/linux/ramips/image/mt7621.mk
sed -i 's/"Phicomm K2P";/"Phicomm K2P (32M)";/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
sed -i 's/<80000000>/<10000000>/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
sed -i 's/m25p,fast-read;/broken-flash-reset;/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
sed -i 's/<0xa0000 0xf60000>/<0xa0000 0x1f60000>/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts

照我的改就可以了。

ltactor, 2021-07-27:
> sed -i 's/15744k/32128k/g' target/linux/ramips/image/mt7621.mk
> sed -i 's/"Phicomm K2P";/"Phicomm K2P (32M)";/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
> sed -i 's/<80000000>/<10000000>/g' [...]

请问为什么降低频率？    感谢回复

ltactor, 2021-07-27:
> 今晚将内核改为了5.4还是不行，自从删除掉32M的K2P选项后，即使将dts和mt7621.mk文件做了相应的改动之后，生成的固件还是没有恢复出厂设置的选项。
>
> 我的刷机步骤是先breed刷入initramfs固件，然后在luci里升级squashfs固件，可是就算在这里升级了squashfs固件重启后，依然没有恢复出厂设置那个按钮。同时系统-软件包里的剩余空间也显示异常，显示剩余167MB [...]

感谢回复，正在使用，我想起来了这是之前的使用32M方法。
非常感谢。

closed 2021-08-15 by AmadeusGhost

ChuJian2, 2022-01-21:
> 感谢回复，正在使用，我想起来了这是之前的使用32M方法。 非常感谢。

> sed -i 's/15744k/32128k/g' target/linux/ramips/image/mt7621.mk sed -i 's/"Phicomm K2P";/"Phicomm K2P (32M)";/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts sed -i 's/<80000000>/<10000000>/g' [...]

您好，请问现在这个您可以正常编译码？我在大概这个issue刚提出来的时候，采用您的方法可以正常编译，没有任何问题，今天我还按照这个来编译，结果先刷ramfs可以正常开机，再刷入sysupgrade后就无法开机了，直接在breed刷sysupgrade也无法正常启动，请问您有相同问题吗？

ltactor, 2022-01-22:
> 
可以，我一直使用这个方法编译。没有问题。