coolsnowwolf/lede #7264: 编译报错

birkh8792, 2021-07-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

编译报错，不知道是不是问题，重新clone了最新的代码编译了一晚上，还是这个错，群主直接把我的issue给close关掉了，任何提示信息也没有，不知道是什么意思？

### (2) 路由器型号和固件版本
x64

### (3) 详细日志

make -C /home/lic/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46 KCFLAGS="-fmacro-prefix-map=/home/lic/lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/lic/lede/staging_dir/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Jul 2 00:20:50 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/lic/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/lic/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V='' cmd_syscalls= KERNELRELEASE=5.10.46 CC="x86_64-openwrt-linux-musl-gcc" bzImage modules
make[5]: Entering directory '/home/lic/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46'
SYNC include/config/auto.conf.cmd
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:480⚠️ multi-line strings not supported
.config:1114⚠️ override: CPU_FREQ_DEFAULT_GOV_SCHEDUTIL changes choice state
.config:3598⚠️ override: M686 changes choice state
.config:6716⚠️ override: SPARSEMEM_MANUAL changes choice state
DESCEND objtool
CALL scripts/atomic/check-atomics.sh
CC arch/x86/kernel/asm-offsets.s
CC /home/lic/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/tools/objtool/elf.o
elf.c: In function 'read_symbols':
elf.c:407:8: error: implicit declaration of function 'gelf_getsymshndx'; did you mean 'gelf_getsym'? [-Werror=implicit-function-declaration]
407 | if (!gelf_getsymshndx(symtab->data, shndx_data, i, &sym->sym,
| ^~~~~~~~~~~~~~~~
| gelf_getsym
CALL scripts/checksyscalls.sh
cc1: all warnings being treated as errors
make[8]: *** [/home/lic/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/tools/build/Makefile.build:96: /home/lic/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/tools/objtool/elf.o] Error 1
make[7]: *** [Makefile:59: /home/lic/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/tools/objtool/objtool-in.o] Error 2
make[6]: *** [Makefile:68: objtool] Error 2
make[5]: *** [Makefile:1947: tools/objtool] Error 2
make[5]: Leaving directory '/home/lic/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46'
make[4]: *** [Makefile:28: /home/lic/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/.modules] Error 2
make[4]: Leaving directory '/home/lic/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/lic/lede/target/linux'
time: target/linux/compile#2.40#0.57#3.07
ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/lic/lede'
make[1]: *** [target/Makefile:18: /home/lic/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/lic/lede'
make: *** [/home/lic/lede/include/toplevel.mk:230：world] 错误 2

minlang112, 2021-07-03:
你动过内核配置？

birkh8792, 2021-07-03:
> 你动过内核配置？

没有

ma20at, 2021-07-03:
make dirclean，deconfig会解决你的问题，今天刚更新编译自己加了一些常用的插件，已经用上最新版本好像上网更快了，哈哈。。。

linxuankent, 2021-07-06:
同样问题

mrclock, 2021-07-08:
我也是这个问题。没动过内核。

mrclock, 2021-07-13:
> make dirclean，deconfig会解决你的问题，今天刚更新编译自己加了一些常用的插件，已经用上最新版本好像上网更快了，哈哈。。。

用make dirclean操作了下，现在编译成功了。

closed 2021-08-23 by AmadeusGhost