coolsnowwolf/lede #7269: 开启DNS缓存后路由器负载很高，导致路由器不可用

slipdaly, 2021-07-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
[image]
DNS解析方式这个是新增的功能吗？开启这个后路由器负载很高，高到最终不可用状态。

### (1) 具体问题

A：开启DNS缓存后路由器负载很高，导致路由器不可用。

### (2) 路由器型号和固件版本

A：Phicomm PSG1218 rev.A

### (3) 详细日志

A：需要什么日志

qqsir-dev, 2021-07-04:
旧版的Turbo ACC里的DNS Caching设置的是内网 AD Guard Home的IP，新版这样就不行，而且不改设置不打开也不行(不能上网)，一定要用公网DNS才可以，这是问题还是变化? 求解？

fuokgfw, 2021-07-04:
这两天把两个加速插件合并之后如果开启dns加速，SSR根本就无法使用，也是蛋疼的，如果在DHCP中取消勾选忽略本地文件就可以，重启之后依旧会自动勾选，没办法只能停用加速了

minlang112, 2021-07-04:
> 如果开启dns加速，SSR根本就无法使用

并未发现“如果开启dns加速，SSR根本就无法使用”的情况，建议详细描述一下问题

hxlls, 2021-07-04:
重新更新下系统，不要保存配置，这个情况我也遇到过，不保存配置更新之后在手动设置一次没有在出现，疑似残留配置引起的故障

fuokgfw, 2021-07-04:
直接重新编译的固件，没保存设置，都遇到这问题，没办法，系统日志中找不到问题，DNS加速直接禁用了就行了

xcf13363175, 2021-07-04:
红米 AX6 编译最新源码固件遇到相同的情况，去掉DNS加速问题解决。

slipdaly, 2021-07-05:
> 重新更新下系统，不要保存配置，这个情况我也遇到过，不保存配置更新之后在手动设置一次没有在出现，疑似残留配置引起的故障

不保存配置，使用breed刷入，问题依旧，目前去掉了DNS加速。

qpzr, 2021-07-05:
> > 重新更新下系统，不要保存配置，这个情况我也遇到过，不保存配置更新之后在手动设置一次没有在出现，疑似残留配置引起的故障
> 
> 不保存配置，使用breed刷入，问题依旧，目前去掉了DNS加速。

我的就更直接了，简单粗暴，哈哈
sed -i 's/+pdnsd-alt/+kmod-fast-classifier +kmod-ipt-offload/g' package/lean/luci-app-turboacc/Makefile

eva2021hd, 2021-08-14:
小娱c5也碰到这个问题。不能上网，luci进不去。家里只能重启。
今天人在碰到这个故障，ssh到路由top查看，发现dnscache占用25%，状态是僵死（z）
8月4日编译

closed 2024-11-19 by coolsnowwolf