coolsnowwolf/lede #7278: watchcat luci界面设置不起作用

ricksuzade-maker, 2021-07-05:
7月5日最新版编译安装后，发现watchcat luci界面设置的参数没有写入到/etc/config/watchcat文件里，文件里还是默认的内容，依然在ping 8.8.8.8。

tone88, 2021-07-05:
我也遇到这个问题。  从5.10内核就开始了。#7271

HoldOnBro, 2021-07-14:
而且这脚本是默认启动的，就很可怕了。如果op没有挂梯子，那每6小时都会被重启一次。

tone88, 2021-07-14:
Wed Jul 14 04:28:30 2021 daemon.info watchcat[9038]: Could not reach 8.8.8.8 for 60. Rebooting after reaching 21600
Wed Jul 14 09:05:32 2021 daemon.info watchcat[9038]: Could not reach 8.8.8.8 for 60. Rebooting after reaching 21600
Wed Jul 14 09:06:32 2021 daemon.info watchcat[9038]: Could not reach 8.8.8.8 for 120. Rebooting after reaching 21600
Wed Jul 14 20:06:34 2021 daemon.info watchcat[9038]: Could not reach 8.8.8.8 for 60. Rebooting after reaching 21600
Wed Jul 14 21:40:34 2021 daemon.info watchcat[9038]: Could not reach 8.8.8.8 for 60. Rebooting after reaching 21600

SSH进配置文件修改后，仍然是ping的8.8.8.8。没修复之前只能先删除这个插件了。 这个插件很冷门吗？望哪位大佬修复下，感谢！！

HoldOnBro, 2021-07-14:
> Wed Jul 14 04:28:30 2021 daemon.info watchcat[9038]: Could not reach 8.8.8.8 for 60. Rebooting after reaching 21600
> Wed Jul 14 09:05:32 2021 daemon.info watchcat[9038]: Could not reach 8.8.8.8 for [...]

修改后在luci点击保存并应用会更新到你手动配置的参数

Tszlam-Cheung, 2021-07-14:
> Wed Jul 14 04:28:30 2021 daemon.info watchcat[9038]: Could not reach 8.8.8.8 for 60. Rebooting after reaching 21600
> Wed Jul 14 09:05:32 2021 daemon.info watchcat[9038]: Could not reach 8.8.8.8 for [...]

ssh进配置修改完IP后 重启一遍就正常了 我也是LUCI改不了 得手动改 不用卸，没事

tone88, 2021-07-15:
> ssh进配置修改完IP后 重启一遍就正常了 我也是LUCI改不了 得手动改 不用卸，没事

那应该是我修改后没重启，所以没生效。  已经卸载了，等大佬修复吧，我感觉这个修复起来应该不困难，但像我们这样的小白就是不会 -_-

82kg, 2021-11-26:
#8304  关issues吧，已经修复了

closed 2021-11-30 by AmadeusGhost