coolsnowwolf/lede #7282: 干净编译出错

qqsir-dev, 2021-07-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
s-1.45.6/.pkgdir/e2fsprogs/usr/lib/
install -d -m0755 /home/qqsir/lede/build_dir/target-x86_64_musl/e2fsprogs-1.45.6/.pkgdir/e2fsprogs/etc/init.d
install -d -m0755 /home/qqsir/lede/build_dir/target-x86_64_musl/e2fsprogs-1.45.6/.pkgdir/e2fsprogs/lib/functions/fsck
install -m0644 ./files/e2fsck.sh /home/qqsir/lede/build_dir/target-x86_64_musl/e2fsprogs-1.45.6/.pkgdir/e2fsprogs/lib/functions/fsck/
install -m0644 ./files/e2fsck.conf /home/qqsir/lede/build_dir/target-x86_64_musl/e2fsprogs-1.45.6/.pkgdir/e2fsprogs/etc/e2fsck.conf
touch /home/qqsir/lede/build_dir/target-x86_64_musl/e2fsprogs-1.45.6/.pkgdir/e2fsprogs.installed
mkdir -p /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '2' | cmp -s - /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libss.version || { echo '2' > /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libss.version;  }
SHELL= flock /home/qqsir/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qqsir/lede/build_dir/target-x86_64_musl/e2fsprogs-1.45.6/.pkgdir/libss/. /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libss_installed
echo "libss" >> /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/e2fsprogs.default.install
mkdir -p /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '2' | cmp -s - /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libext2fs.version || { echo '2' > /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libext2fs.version;  }
SHELL= flock /home/qqsir/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qqsir/lede/build_dir/target-x86_64_musl/e2fsprogs-1.45.6/.pkgdir/libext2fs/. /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libext2fs_installed
echo "libext2fs" >> /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/e2fsprogs.default.install
mkdir -p /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/qqsir/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qqsir/lede/build_dir/target-x86_64_musl/e2fsprogs-1.45.6/.pkgdir/e2fsprogs/. /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.e2fsprogs_installed
echo "e2fsprogs" >> /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/e2fsprogs.default.install
make[3]: Leaving directory '/home/qqsir/lede/package/utils/e2fsprogs'
time: package/utils/e2fsprogs/compile#43.84#6.53#71.37
make[2]: Leaving directory '/home/qqsir/lede'
make[1]: *** [package/Makefile:110: /home/qqsir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qqsir/lede'
make: *** [/home/qqsir/lede/include/toplevel.mk:230: world] Error 2
qqsir@code:~/lede$

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

EnnawYang, 2021-07-05:
无有效日志，另外你怎么老是出错？

shoucandanghehe, 2021-07-05:
同样的错误
用actions编译的
```
2021-07-05T12:15:50.5665428Z *** Installing runtime programs
2021-07-05T12:15:50.5682106Z install apps/openssl -> /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/usr/bin/openssl
2021-07-05T12:15:50.5740190Z install ./tools/c_rehash -> /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/usr/bin/c_rehash
2021-07-05T12:15:50.5802381Z created directory `/home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc'
2021-07-05T12:15:50.5804908Z created directory `/home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl'
2021-07-05T12:15:50.5807646Z created directory `/home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/certs'
2021-07-05T12:15:50.5837448Z created directory `/home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/private'
2021-07-05T12:15:50.5866845Z created directory `/home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/misc'
2021-07-05T12:15:50.5910186Z install ./apps/CA.pl -> /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/misc/CA.pl
2021-07-05T12:15:50.5971993Z install ./apps/tsget.pl -> /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/misc/tsget.pl
2021-07-05T12:15:50.6014609Z link /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/misc/tsget -> /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/misc/tsget.pl
2021-07-05T12:15:50.6029019Z install ./apps/openssl.cnf -> /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/openssl.cnf.dist
2021-07-05T12:15:50.6090342Z install ./apps/openssl.cnf -> /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/openssl.cnf
2021-07-05T12:15:50.6122156Z install ./apps/ct_log_list.cnf -> /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/ct_log_list.cnf.dist
2021-07-05T12:15:50.6179501Z install ./apps/ct_log_list.cnf -> /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/etc/ssl/ct_log_list.cnf
2021-07-05T12:15:50.6199495Z make[4]: Leaving directory '/home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k'
2021-07-05T12:15:50.6201542Z touch /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.built
2021-07-05T12:15:50.6234240Z rm -rf /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl.installed /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl
2021-07-05T12:15:50.6257310Z mkdir -p /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl
2021-07-05T12:15:50.6287644Z install -d -m0755 /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl/etc/ssl/certs
2021-07-05T12:15:50.6316889Z install -d -m0755 /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl/etc/ssl/private
2021-07-05T12:15:50.6349053Z chmod 0700 /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl/etc/ssl/private
2021-07-05T12:15:50.6372256Z install -d -m0755 /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl/usr/lib
2021-07-05T12:15:50.6403729Z install -m0644 /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/usr/lib/libcrypto.so.* /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl/usr/lib/
2021-07-05T12:15:50.6466340Z install -m0644 /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/ipkg-install/usr/lib/libssl.so.* /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl/usr/lib/
2021-07-05T12:15:50.6505945Z install -d -m0755 /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl/usr/lib/engines-1.1
2021-07-05T12:15:50.6533951Z touch /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/build_dir/target-x86_64_musl/openssl-1.1.1k/.pkgdir/libopenssl.installed
2021-07-05T12:15:50.6560614Z mkdir -p /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/staging_dir/target-x86_64_musl/root-x86/stamp
2021-07-05T12:15:50.6592705Z echo '1.1' | cmp -s - /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/staging_dir/target-x86_64_musl/pkginfo/libopenssl.version || { echo '1.1' > /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/staging_dir/target-x86_64_musl/pkginfo/libopenssl.version;  } 
[... cut ...]
2021-07-05T12:15:51.6497597Z make[2]: Leaving directory '/home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede'
2021-07-05T12:15:51.6514724Z make[1]: *** [package/Makefile:110: /home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2021-07-05T12:15:51.6516710Z make[1]: Leaving directory '/home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede'
2021-07-05T12:15:51.6521955Z make: *** [/home/runner/work/Auto-OpenWrt/Auto-OpenWrt/lede/include/toplevel.mk:230: world] Error 2
2021-07-05T12:15:51.6620072Z ##[error]Process completed with exit code 2.
```

qqsir-dev, 2021-07-05:
> 无有效日志，另外你怎么老是出错？

因为我编译的频率高
还有，我要是知道我怎么老是出错我还用在这问吗？你问题逻辑有问题
我能做的，ubuntu有更新我就先更新系统直到没有任何更新
删除lede重头来，还是这个错误，还能怎么办？

发错误又不是捣乱，一是寻求帮助，二是看看是不是确实哪里有问题
日志有没有效果我也不清楚，因为我除了按照官网的要求的步骤去编译有以外什么都不懂，我不知道什么是有效日志。你如果懂就应该指出怎么才能提供有效的日志，而不是只会说无效，然后还说我总出错。

EnnawYang, 2021-07-05:
>> 无有效日志，另外你怎么老是出错？
>
> 你如果懂就应该指出怎么才能提供有效的日志，而不是只会说无效，然后还说我总出错。
> 
1. 请用-j1跑一遍，把编译出错的包从编译开始到编译停止部分的日志完整贴出来（3楼也一样）  
2. 我这后半句不是疑问句

qqsir-dev, 2021-07-05:
> > > 无有效日志，另外你怎么老是出错？
> > 
> > 
> > 你如果懂就应该指出怎么才能提供有效的日志，而不是只会说无效，然后还说我总出错。
> 
> 1. 请用-j1跑一遍，把编译出错的包从编译开始到编译停止部分的日志完整贴出来（3楼也一样）
> 2. 我这后半句不是疑问句

好的，以后出错我会尝试你这个方法，30秒前刚刚编译成功了。

closed 2021-07-05 by qqsir-dev

zwillhill, 2021-07-08:
> > > > 无有效日志，另外你怎么老是出错？
> > > 
> > > 
> > > 你如果懂就应该指出怎么才能提供有效的日志，而不是只会说无效，然后还说我总出错。
> > 
> > 
> > 
> > 1. 请用-j1跑一遍，把编译出错的包从编译开始到编译停止部分的日志完整贴出来（3楼也一样）
> > 2. 我这后半句不是疑问句
> 
> 好的，以后出错我会尝试你这个方法，30秒前刚刚编译成功了。

本地编译时如果出错，控制台就会提示用-j1参数才能显示具体错误内容。
actions用-j1或-j4编译，粗略测试，好像花费时间差不多