coolsnowwolf/lede #7283: 今天升级代码后，出现一个莫名其妙的问题

gswangmd-gssi, 2021-07-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
单网口，做旁路由器使用，昨天以前每次升级都没问题，今天升级编译后，正常运行一两分钟后就无法解析域名，打不开网页。

kisaragychihaya, 2021-07-05:
日志和型号都没有，看个鬼

gswangmd-gssi, 2021-07-05:
刚才测试了一下，停了dnsfilter后就正常了。
Mon Jul  5 21:43:30 2021 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Mon Jul  5 21:44:45 2021 cron.err crond[9080]: crond (busybox 1.33.1) started, log level 5
Mon Jul  5 21:44:45 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul  5 21:44:45 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul  5 21:44:45 2021 daemon.crit dnsmasq[9225]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:44:45 2021 daemon.crit dnsmasq[9225]: FAILED to start up
Mon Jul  5 21:44:50 2021 daemon.crit dnsmasq[9257]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:44:50 2021 daemon.crit dnsmasq[9257]: FAILED to start up
Mon Jul  5 21:44:55 2021 daemon.crit dnsmasq[9268]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:44:55 2021 daemon.crit dnsmasq[9268]: FAILED to start up
Mon Jul  5 21:45:00 2021 daemon.crit dnsmasq[9279]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:45:00 2021 daemon.crit dnsmasq[9279]: FAILED to start up
Mon Jul  5 21:45:05 2021 daemon.crit dnsmasq[9295]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:45:05 2021 daemon.crit dnsmasq[9295]: FAILED to start up
Mon Jul  5 21:45:10 2021 daemon.crit dnsmasq[9318]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:45:10 2021 daemon.crit dnsmasq[9318]: FAILED to start up
Mon Jul  5 21:45:10 2021 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Mon Jul  5 21:45:54 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul  5 21:45:54 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul  5 21:45:54 2021 daemon.crit dnsmasq[9687]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:45:54 2021 daemon.crit dnsmasq[9687]: FAILED to start up
Mon Jul  5 21:45:59 2021 daemon.crit dnsmasq[9727]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:45:59 2021 daemon.crit dnsmasq[9727]: FAILED to start up
Mon Jul  5 21:46:04 2021 daemon.crit dnsmasq[9765]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:46:04 2021 daemon.crit dnsmasq[9765]: FAILED to start up
Mon Jul  5 21:46:09 2021 daemon.crit dnsmasq[9788]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:46:09 2021 daemon.crit dnsmasq[9788]: FAILED to start up
Mon Jul  5 21:46:14 2021 daemon.crit dnsmasq[9799]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:46:14 2021 daemon.crit dnsmasq[9799]: FAILED to start up
Mon Jul  5 21:46:19 2021 daemon.crit dnsmasq[9836]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:46:19 2021 daemon.crit dnsmasq[9836]: FAILED to start up
Mon Jul  5 21:46:19 2021 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Mon Jul  5 21:46:49 2021 daemon.err uhttpd[4072]: cat: write error: Broken pipe
Mon Jul  5 21:47:32 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul  5 21:47:32 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul  5 21:47:32 2021 daemon.crit dnsmasq[10812]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:47:32 2021 daemon.crit dnsmasq[10812]: FAILED to start up
Mon Jul  5 21:47:37 2021 daemon.crit dnsmasq[10876]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:47:37 2021 daemon.crit dnsmasq[10876]: FAILED to start up
Mon Jul  5 21:47:42 2021 daemon.crit dnsmasq[10918]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:47:42 2021 daemon.crit dnsmasq[10918]: FAILED to start up
Mon Jul  5 21:47:47 2021 daemon.crit dnsmasq[10953]: bad option at line 1 of /tmp/dnsmasq.dnsfilter/rules.conf
Mon Jul  5 21:47:47 2021 daemon.crit dnsmasq[10953]: FAILED to start up
Mon Jul  5 21:47:49 2021 cron.err crond[11080]: crond (busybox 1.33.1) started, log level 5
Mon Jul  5 21:47:49 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jul  5 21:47:49 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jul  5 21:47:49 2021 daemon.info dnsmasq[11226]: Connected to system UBus

closed 2021-07-05 by gswangmd-gssi