coolsnowwolf/lede #7285: 编译新版本报错 error: implicit declaration of function 'gelf_getsymshndx'; did you mean 'gelf_getsym'? [-Werror=implicit-function-declaration]

linxuankent, 2021-07-05:
make[5]: Entering directory '/home/linxuan/snap/ceshi/test/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  DESCEND  objtool
  CC       /home/linxuan/snap/ceshi/test/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/tools/objtool/elf.o
elf.c: In function 'read_symbols':
elf.c:407:8: error: implicit declaration of function 'gelf_getsymshndx'; did you mean 'gelf_getsym'? [-Werror=implicit-function-declaration]
  407 |   if (!gelf_getsymshndx(symtab->data, shndx_data, i, &sym->sym,
      |        ^~~~~~~~~~~~~~~~
      |        gelf_getsym
cc1: all warnings being treated as errors
make[8]: *** [/home/linxuan/snap/ceshi/test/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/tools/build/Makefile.build:97: /home/linxuan/snap/ceshi/test/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/tools/objtool/elf.o] Error 1
make[7]: *** [Makefile:59: /home/linxuan/snap/ceshi/test/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/tools/objtool/objtool-in.o] Error 2
make[6]: *** [Makefile:68: objtool] Error 2
make[5]: *** [Makefile:1947: tools/objtool] Error 2
make[5]: Leaving directory '/home/linxuan/snap/ceshi/test/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46'
make[4]: *** [Makefile:28: /home/linxuan/snap/ceshi/test/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.46/.modules] Error 2
make[4]: Leaving directory '/home/linxuan/snap/ceshi/test/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/linxuan/snap/ceshi/test/lede/target/linux'
time: target/linux/compile#0.87#0.18#1.03
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/linxuan/snap/ceshi/test/lede'
make[1]: *** [target/Makefile:18: /home/linxuan/snap/ceshi/test/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/linxuan/snap/ceshi/test/lede'
make: *** [/home/linxuan/snap/ceshi/test/lede/include/toplevel.mk:230：world] 错误 2

linxuankent, 2021-07-06:
这是什么原因？

linxuankent, 2021-07-06:
这是什么原因？

whales-lee, 2021-09-18:
Your build tree is tainted and contains old libelf residues after bf4aa0c6a2f ("tools/libelf: remove unneeded host library"). That old libelf version doesn't work with kernels 5.8+ https://lore.kernel.org/all/20200421220843.188260-2-samitolvanen@google.com/T/#u so you either need to cleanup your build tree manually (`find staging_dir/host -name *libelf* -delete`), build from scratch or perform `make distclean`.
https://bugs.openwrt.org/index.php?do=details&task_id=3707