coolsnowwolf/lede #7288: qBittorrent-static 编译失败

naoki66, 2021-07-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：qBittorrent-static编译失败

### (2) 路由器型号和固件版本

A：x64

### (3) 详细日志

A：Collected errors:
 * check_data_file_clashes: Package qBittorrent-static wants to install file /home/XXXX/lede/build_dir/target-x86_64_musl/root-x86/usr/bin/qbittorrent-nox
        But that file is already provided by package  * qbittorrent
 * opkg_install_cmd: Cannot install package qBittorrent-static.
package/Makefile:66: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/alex/lede'
package/Makefile:111: recipe for target '/home/alex/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/alex/XXXX/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2

naoki66, 2021-07-06:
qbittorrent 包名错了？

coolsnowwolf, 2021-07-06:
重复了

myannnascar, 2021-07-06:
> 重复了

这个问题该如何解决呢哥

coolsnowwolf, 2021-07-06:
不要同时编译2个 qbittorrent

naoki66, 2021-07-06:
主要是菜单选择普通版，但是NETWORK-bittorrent会自动被选择。
导致普通版本无法编译

PandaEgg, 2021-07-06:
如何取消选择呢？

openwrt-try, 2021-07-07:
network->bittorrent,
取消勾选qbittorrent或qbittorrent-static

closed 2021-08-15 by AmadeusGhost