coolsnowwolf/lede #7296: AX6软转发效率不高，且CPU不满载

xsm1997, 2021-07-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Redmi AX6路由器在软转发时，效率不高，只能跑30MB/s左右。但奇怪的是，此时CPU并没有满载。使用htop看每个核心的负载时，核心0的负载较高，但没有满载。尝试换过多种go编写的软转发程序和rust编写的软转发程序，均有此问题，且达到的带宽基本相同（都是30M-35M）。相近时间使用x86_64软路由运行上述程序则没有问题（能达到60M-70M）。

是此路由软转发只能达到这个带宽，还是其他哪里有问题？

### (2) 路由器型号和固件版本

A：Redmi AX6
430a7b4

### (3) 详细日志

A：
[image]
[image]
[system.txt](https://github.com/coolsnowwolf/lede/files/6770572/system.txt)
[kernel.txt](https://github.com/coolsnowwolf/lede/files/6770573/kernel.txt)

ldy8182, 2021-07-08:
ax6用nss不香吗？

xsm1997, 2021-07-10:
找到问题了。

软转发效率不高实际上是因为网卡的sirq没有负载均衡，都作用在一个核心上。开启网卡的rps和xps功能能解决问题。
开启后的某软件内网服务器运行效率如图所示，图1为转发https流量（不二次加密），图2为转发http流量（使用aes加密，有硬件加速）的多线程流量转发速度。转发https流量不进行二次加密的情况下，已经可以近似认为跑满千兆。~~（预感到AX6马上涨价~~

[image]
[image]

附上临时开启网卡rps和xps的命令，可以添加到防火墙自定义规则或是其他能自动运行的地方：
```
for q in $(ls /sys/class/net/eth*/queues/rx-*/rps_cpus); do echo f > $q; done
for q in $(ls /sys/class/net/eth*/queues/rx-*/rps_flow_cnt); do echo 4096 > $q; done
for q in $(ls /sys/class/net/eth*/queues/tx-*/xps_cpus); do echo f > $q; done
echo 32768 > /proc/sys/net/core/rps_sock_flow_entries
```

恳请lean大雕有时间给网卡默认开启rps和xps，或者添加到启动脚本。 @coolsnowwolf

huaihuai0406, 2021-07-14:
同遇到这个问题 直插LAN口作ap LAN to LAN的速率只能在300-400Mbps 按楼上开启rps xps 之后可以跑满千兆 NSS 下cpu占应用 10%以内 @coolsnowwolf

xsm1997, 2021-07-14:
@huaihuai0406 请问您有遇到开了nss-ecm之后随机重启的问题吗？是如何解决的呢？

ldy8182, 2021-07-15:
按照楼上开启rps xps 之后
[image]

huaihuai0406, 2021-07-15:
@xsm1997 我没有遇到随机重启的问题 试试清空缓存重新编译

xsm1997, 2021-07-15:
@huaihuai0406 
1 您有在使用nss-ecm模块吗？我加上这个模块，就随机重启，不开启这个，但开启其他nss模块，也不会重启。
2 另外您有在使用pppoe吗？我编译nss-pppoe模块之后，断开或重连pppoe连接会直接重启......
3 还有，您在使用luci-app-turboacc吗？编译了sfe或flow offload吗？

huaihuai0406, 2021-07-15:
@xsm1997 
1. 有编译nss-ecm 这个应该是默认配置了
2. 现在是ap模式使用没有测试pppoe拨号 回去我测试下
3. 有使用，编译了sfe
我编译的固件和配置 https://wwe.lanzoui.com/ih2rGrgf2je

coolsnowwolf, 2021-07-16:
已添加，请编译测试

ldy8182, 2021-07-16:
我测试了，没遇到问题

xsm1997, 2021-07-18:
测试过了，没问题。

xsm1997, 2021-07-18:
@huaihuai0406 @coolsnowwolf 
干净编译之后不随机重启了，但是发现极限转发速度的话，用SFE会比用nss-ecm快一些（sfe有90M，nss只有80M），不知道是哪里出现的问题。按理说nss不应该比sfe还慢。

closed 2021-07-18 by xsm1997