coolsnowwolf/lede #7300: 8125网卡驱动高负载场景下会断流

MrEsquire, 2021-07-06:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
网卡高负载场景下,本地客户端电脑会直接断网,显示无网络链接,大概5秒左右会恢复
本地客户端和路由器都是用的8125网卡,此问题最近才出现.
**以下是下载时流量突然归零,网络显示未连接的截图**
[image]

### (1) 具体问题

A：8125网卡断流

### (2) 路由器型号和固件版本

A：系统:OpenWrt R21.7.1.100 / LuCI Master (git-21.181.13804-f7b5f6d)
      宿主配置:esxi 7.0 Update 2
                     i5-10600
                     i350t4
                     8125
      路由配置:i350 x3(直通)
                     
### (3) 详细日志

A：

EnnawYang, 2021-07-07:
我的是8125b，没有发现这个问题，可以看下你的bios设置 `ASPM` 有没有关闭

MrEsquire, 2021-07-07:
> 我的是8125b，没有发现这个问题，可以看下你的bios设置 `ASPM` 有没有关闭

今晚我回家再排查下

closed 2021-07-07 by MrEsquire