coolsnowwolf/lede #7302: 额外的 HOSTS 文件配置不生效

slipdalysmall, 2021-07-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：额外的 HOSTS 文件配置不生效，配置完成后ping github.com没映射到配置文件中的地址，将配置文件内容添加到/etc/hosts中是生效的。

[image]
[image]

### (2) 路由器型号和固件版本

A：
型号：Phicomm PSG1218 rev.A
固件版本：OpenWrt R21.7.1.100 / LuCI Master (git-21.181.13804-f7b5f6d)
内核版本 ：5.4.128

### (3) 详细日志

A：
/etc/init.d/dnsmasq restart返回udhcpc: no lease, failing
[image]

hxlls, 2021-07-07:
[image]
实测没问题，建议自己检查

slipdaly, 2021-07-07:
> [image]
> 实测没问题，建议自己检查，图片上应该能看明白

是我使用的Adbyby Plus+ 有冲突导致的么，我这里确实是不行

slipdalysmall, 2021-07-07:
> [image]
> 实测没问题，建议自己检查，图片上应该能看明白

是我使用的Adbyby Plus+ 有冲突导致的么，我这里确实是不行

slipdalysmall, 2021-07-07:
> [image]
> 实测没问题，建议自己检查，图片上应该能看明白

找到问题了，文件权限没对，权限改为644就OK了，感谢

closed 2021-07-07 by slipdalysmall