coolsnowwolf/lede #7304: 新版本的turboacc，解决了使用SFE导致路由器无规律重启的问题吗

NagaseKouichi, 2021-07-07:
之前的turboacc开启SFE的话，会随机导致路由器重启，想问下各位大佬这个问题新版的turboacc解决了吗？

OldCoding, 2021-07-07:
无规律重启？什么设备？从没遇到过这种情况

rongcenter, 2021-07-07:
一直都用的sfe，之前看dadiao调查过，不过我没出现过。

keeno1982, 2021-07-09:
我也是开SFE就无规律自动重启，设备N1和树莓派3b都这样，不知道现在有没有改善

Kaze1027, 2021-07-12:
一样的问题，我发现是更新了add NSS ecm 64 support之后出现的问题，之前编译的那个固件很稳定，我在怀疑是不是nss有bug

keeno1982, 2021-07-13:
用了新版的acc，一天多还没遇到自动重启，看来是有所改善，以前一天都要抽好几次

Tszlam-Cheung, 2021-07-13:
> 我也是开SFE就无规律自动重启，设备N1和树莓派3b都这样，不知道现在有没有改善

嗯 对 测试了我之前是G4900+ H310M 之前SFE无规律自动重启的问题我遇到了

20210704编译了下 试了新版的acc的sfe
新的倒是不重启了 不断网了
但是会随机ping+100ms 默认配置 CPU governor 是powersave mode
所以现在把governor设成performance了 然后关掉SFE 一切正常

Kaze1027, 2021-07-14:
我现在ax6退回到7月9日的没有添加nss的版本了，稳定运行不跳ping不重启，堪称完美

Kaze1027, 2021-07-19:
我发现了带nss版本的固件，如果设备除了wan以外不接任何其他有线可以有效缓解这个问题

zhouxiaolong920, 2021-08-06:
就不能，不用这个二个插件吗？turboacc和short fe？

x86的是性能不够？

closed 2022-02-12 by Beginner-Go