coolsnowwolf/lede #7313: 系统日志错误

deplives, 2021-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
系统日志中频繁出现
```
kern.err kernel: [   39.245084] squashfs: Unknown parameter 'discard'
kern.info kernel: [   39.247411] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
kern.err kernel: [   39.249164] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
kern.info kernel: [   39.251095] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x80a345bd)
kern.err kernel: [   39.252659] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
kern.err kernel: [   39.255796] exFAT-fs (sda128): invalid boot record signature
kern.err kernel: [   39.257026] exFAT-fs (sda128): failed to read boot sector
kern.err kernel: [   39.258192] exFAT-fs (sda128): failed to recognize exfat type
```

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：

aazhe1984, 2022-04-19:
I have same  issue.

XCDream, 2022-05-16:
i have the same problem too, but the system still works with no crash or other errors.i have no idea about it and because i dont know whats the problem is, so i just ignore it.

closed 2024-11-19 by coolsnowwolf