coolsnowwolf/lede #7316: 日志中出现bash: line 6: 8: Bad file descriptor，但不影响编译固件

adamgonglj, 2021-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：
R2S和R4S都有这样的问题。
### (3) 详细日志

A：
298+1 records in
298+1 records out
153051 bytes (153 kB, 149 KiB) copied, 0.00241204 s, 63.5 MB/s
1738+1 records in
1738+1 records out
890272 bytes (890 kB, 869 KiB) copied, 0.0067687 s, 132 MB/s
bash: line 6: 8: Bad file descriptor

adamgonglj, 2021-07-08:
就是最后一行。。。有这样的问题

keeno1982, 2021-07-08:
同样遇到这个问题，如何解决？
Generating index for package ./rpcd-mod-rrdns_20170710_aarch64_cortex-a53.ipk
bash: line 6: 8: Bad file descriptor
Generating index for package ./boost-date_time_1.76.0-1_aarch64_cortex-a53.ipk

adamgonglj, 2021-07-09:
我现在不知道怎么解决。

AmadeusGhost, 2021-07-09:
忽略即可，无任何影响

keeno1982, 2021-07-09:
强迫症不能忍

adamgonglj, 2021-07-11:
> 忽略即可，无任何影响

好的。

closed 2021-07-11 by adamgonglj