coolsnowwolf/lede #7318: 能否添加紅米AX5/小米AX1800的支援

rain931215, 2021-07-08:
CPU : 高通IPQ6018 4核Cortex-A53
交換器芯片: 高通QCA8075 eth1~3為lan eth4為wan
256M記憶體
128M NAND
AX5應該跟AX6相似只是CPU從IPQ8071A變成IPQ6018還有規格縮水
5g速率砍半

底下這兩個專案好像有IPQ6018支援
https://github.com/Telecominfraproject/wlan-ap
https://github.com/8devices/openwrt-8devices/tree/qsdk-11.3-ipq60xx-4.4
稍微看了下第一個直接把ipq6018歸類在ipq807x的subtarget
猜測6018跟8071A應該是差不多的

AX5與AX1800看https://www.acwifi.net/10272.html 的拆解兩者是一樣的硬體

AX5分區表:
```
dev:    size   erasesize  name
mtd0: 00180000 00020000 "0:SBL1"
mtd1: 00100000 00020000 "0:MIBIB"
mtd2: 00380000 00020000 "0:QSEE"
mtd3: 00080000 00020000 "0:DEVCFG"
mtd4: 00080000 00020000 "0:RPM"
mtd5: 00080000 00020000 "0:CDT"
mtd6: 00080000 00020000 "0:APPSBLENV"
mtd7: 00180000 00020000 "0:APPSBL"
mtd8: 00080000 00020000 "0:ART"
mtd9: 00080000 00020000 "bdata"
mtd10: 00080000 00020000 "crash"
mtd11: 00080000 00020000 "crash_syslog"
mtd12: 00080000 00020000 "0:BOOTCONFIG"
mtd13: 00080000 00020000 "0:BOOTCONFIG1"
mtd14: 00380000 00020000 "0:QSEE_1"
mtd15: 00080000 00020000 "0:DEVCFG_1"
mtd16: 00080000 00020000 "0:RPM_1"
mtd17: 00080000 00020000 "0:CDT_1"
mtd18: 02400000 00020000 "rootfs"
mtd19: 02400000 00020000 "rootfs_1"
mtd20: 024a0000 00020000 "overlay"
mtd21: 00080000 00020000 "cfg_bak"
mtd22: 003a2000 0001f000 "kernel"
mtd23: 0139e000 0001f000 "ubi_rootfs"
mtd24: 008b8000 0001f000 "rootfs_data"
mtd25: 02093000 0001f000 "data"
```

xfmly, 2021-07-09:
ipq6000，CPU性能强于ipq8071，能刷openwrt做主路由就强了。

AmadeusGhost, 2021-07-09:
> ipq6000，CPU性能强于ipq8071，能刷openwrt做主路由就强了。

你确定？

xfmly, 2021-07-09:
[image]
大佬，看资料得来的。

AmadeusGhost, 2021-07-09:
> [image]
> 大佬，看资料得来的。

这垃圾图别看了，数据根本不准，连最基本的变量都没有控制好。

xfmly, 2021-07-09:
> > [image]
> > 大佬，看资料得来的。
> 
> 这垃圾图别调查，数据根本不准，连最根本的原因都没有控制好。

知道了，谢谢大佬！

lsyyy233, 2021-07-09:
> [image]
> 大佬，看资料得来的。

1G的A53单核性能是1.5G的A72的两倍就过分了

rsd1987, 2021-07-12:
他这个跑的是aes 128 gcm  刚好树莓派4b没有硬件加速，所以很垃圾

closed 2021-08-15 by AmadeusGhost

PhamtomK12, 2021-09-28:
L大什么时候搞一搞啊 家里的AX1800都吃灰了