coolsnowwolf/lede #7326: 编译出问题了.

gzw13999, 2021-07-09:
checking for mkdtemp... yes
checking for mkfifo... yes
checking whether mkfifo rejects trailing slashes... yes
checking whether mknod can create fifo without root privileges... configure: error: in `/root/lede/build_dir/host/tar-1.32':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make[3]: *** [Makefile:32: /root/lede/build_dir/host/tar-1.32/.configured] Error 1
make[3]: Leaving directory '/root/lede/tools/tar'
time: tools/tar/compile#10.15#4.06#17.11
    ERROR: tools/tar failed to build.
make[2]: *** [tools/Makefile:159: tools/tar/compile] Error 1
make[2]: Leaving directory '/root/lede'
make[1]: *** [tools/Makefile:155: /root/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/root/lede'
make: *** [/root/lede/include/toplevel.mk:230: world] Error 2
[1]+  Exit 2                  nohup make -j1 V=s

hxlls, 2021-07-09:
用root 编译？最基本的readme 都不看？

hyird, 2021-07-09:
export FORCE_UNSAFE_CONFIGURE=1

closed 2021-07-10 by coolsnowwolf