coolsnowwolf/lede #7332: 新版5.10的NFS共享不工作，求助

guochan2019, 2021-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
本人x86软路由，昨天按照L大的提示全新拉取代码编译5.10，带配置从5.4升级，使用过程中发现NFS管理中共享的目录均无法打开，排查组件与5.4时也没有出入，不知是什么原因，有知情的大佬没

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

fshh1988, 2021-07-10:
我也碰到同样问题，只要是挂载过的目录都无法访问，但是你如果共享/etc目录，是可以访问的

3mile, 2021-10-19:
同样的问题。
看了外网有回答说是不能挂载/mnt下。所以用
mount --bind /mnt/sda3 /tmp/share/sda3
然后/tmp/share/sda3       192.168.1.0/24(rw,sync,root_squash,all_squash,insecure,no_subtree_check)
客户端访问就会卡死

fshh1988, 2021-10-30:
问题解决了，因为现在编译的固件只带NFSv4版本，v4的共享要这样配置：
/mnt/sda1	*(rw,all_squash,insecure,no_subtree_check,sync,fsid=1)
/mnt/sda2	*(rw,all_squash,insecure,no_subtree_check,sync,fsid=2)
/mnt/sdb1	*(rw,all_squash,insecure,no_subtree_check,sync,fsid=3)
/mnt/sdb2	*(rw,all_squash,insecure,no_subtree_check,sync,fsid=4)
/mnt/sdb3	*(rw,all_squash,insecure,no_subtree_check,sync,fsid=5)
/mnt/sdb4	*(rw,all_squash,insecure,no_subtree_check,sync,fsid=6)
注意最后的这个参数，fsid，每个共享的值是不一样的，如果一样就会导致访问不了 @3mile @guochan2019

guochan2019, 2021-10-30:
> rw,all_squash,insecure,no_subtree_check,sync,fsid=1

真的解决了，非常感谢兄弟研究并解决

3mile, 2021-11-10:
多谢@fshh1988兄，原来是启用了nfsv4的原因。

closed 2024-11-19 by coolsnowwolf