coolsnowwolf/lede #7333: Haproxy 编译完无法运行

zfh18, 2021-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Haproxy 无法运行

### (2) 路由器型号和固件版本

A：ESXI x86 OpenWrt R21.7.1.100 / LuCI Master (git-21.188.53404-317d4ad)

### (3) 详细日志

A：Sat Jul 10 13:49:35 2021 user.notice haproxy: is initializing enabled is 1
Sat Jul 10 13:49:36 2021 user.notice restarting: haproxy
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy: Syntax: /etc/init.d/haproxy [command]
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy:
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy: Available commands:
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy: 	start           Start the service
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy: 	stop            Stop the service
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy: 	restart         Restart the service
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy: 	reload          Reload configuration files (or restart if service does not implement reload)
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy: 	enable          Enable service autostart
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy: 	disable         Disable service autostart
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy: 	enabled         Check if service is started on boot
Sat Jul 10 13:49:36 2021 daemon.notice procd: /etc/init.d/haproxy:
Sat Jul 10 13:50:02 2021 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

skydevil88, 2021-07-11:
一样问题，6月初编译还能正常运行。

Beginner-Go, 2021-07-15:
https://github.com/coolsnowwolf/packages/pull/191
can you test?

redjumper, 2021-07-19:
7.19最新编译的也是无法正常运行
daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

redjumper, 2021-08-12:
mode health在新版haproxy已弃用，修改/etc/haproxy.cfg，找到mode health，改成http-request return status 200试试

closed 2022-02-12 by Beginner-Go