coolsnowwolf/lede #7337: 请教各位大佬，关于K2P，Wi-Fi驱动

qjiang888, 2021-07-10:
OpenWrt R21.6.1 / LuCI Master (git-21.175.10795-3d7bad0)  这个版本的wifi一点问题都没有，今天编译了最新版又变成去年那个mtkwifi了，还带中继，这个中继驱动非常难用，启用或者停用Wi-Fi 都会卡很久，经常500错误，刷完机第一次不会自动启动无线，小白想问下可以自行换回原来的wifi驱动吗？

qjiang888, 2021-07-10:
500 内部服务器错误
对不起，服务器遇到未知错误。
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute template dispatcher target for entry '/admin/network/wifi'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'admin_mtk/mtk_wifi_overview'.
A runtime error occured: /usr/lib/lua/mtkwifi.lua:491: Interrupted system call
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function

lphgor, 2021-07-10:
把这一行：
https://github.com/coolsnowwolf/lede/blob/aa73053a71533ce5e3f4b2ab975a5657a6e033fa/target/linux/ramips/image/mt7621.mk#L1078
改回原来的试试：
https://github.com/coolsnowwolf/lede/blob/213f55ba583ab1c41d4f58d8f257b8c0e205d4f8/target/linux/ramips/image/mt7621.mk#L844

qjiang888, 2021-07-11:
> ：
> 
> https://github.com/coolsnowwolf/lede/blob/aa73053a71533ce5e3f4b2ab975a5657a6e033fa/target/linux/ramips/image/mt7621.mk#L1078
> 
> 
> 试试原来的：
> https://github.com/coolsnowwolf/lede/blob/213f55ba583ab1c41d4f58d8f257b8c0e205d4f8/target/linux/ramips/image/mt7621.mk#L844

非常感谢大佬，K2P终于有救了！，那个中继驱动是真的有问题。

closed 2021-08-15 by AmadeusGhost