coolsnowwolf/lede #7341: qca-nss编译报错

aa4851698, 2021-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用的默认选项

### (2) 路由器型号和固件版本

A：锐捷m520

### (3) 详细日志

A：icit-function-declaration]
  if (nf_conntrack_dscpremark_ext_get_dscp_rule_validity(ct)
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      nf_conntrack_dscpremark_ext_init
cc1: all warnings being treated as errors
scripts/Makefile.build:261: recipe for target '/home/cxx/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-nss-ecm-noload/qca-nss-ecm-9228212b/ecm_classifier_dscp.o' failed
make[5]: *** [/home/cxx/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-nss-ecm-noload/qca-nss-ecm-9228212b/ecm_classifier_dscp.o] Error 1
Makefile:1734: recipe for target '/home/cxx/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-nss-ecm-noload/qca-nss-ecm-9228212b' failed
make[4]: *** [/home/cxx/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-nss-ecm-noload/qca-nss-ecm-9228212b] Error 2
make[4]: Leaving directory '/home/cxx/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.4.128'
Makefile:270: recipe for target '/home/cxx/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-nss-ecm-noload/qca-nss-ecm-9228212b/.built' failed
make[3]: *** [/home/cxx/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-nss-ecm-noload/qca-nss-ecm-9228212b/.built] Error 2
make[3]: Leaving directory '/home/cxx/lede/package/nss/qca/qca-nss-ecm'
time: package/nss/qca/qca-nss-ecm/noload/compile#1.34#0.60#2.37
    ERROR: package/nss/qca/qca-nss-ecm failed to build (build variant: noload).
package/Makefile:114: recipe for target 'package/nss/qca/qca-nss-ecm/compile' failed
make[2]: *** [package/nss/qca/qca-nss-ecm/compile] Error 1
make[2]: Leaving directory '/home/cxx/lede'
package/Makefile:110: recipe for target '/home/cxx/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/cxx/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/cxx/lede'
/home/cxx/lede/include/toplevel.mk:228: recipe for target 'world' failed

AmadeusGhost, 2021-07-12:
> scripts/Makefile.build:261: recipe for target '/home/cxx/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-nss-ecm-noload/qca-nss-ecm-9228212b/ecm_classifier_dscp.o' failed

nss 默认选上了，不要再去选其它的

closed 2021-07-13 by coolsnowwolf

xdcabc, 2022-09-17:
请问解决了吗？同款路由器同样问题