coolsnowwolf/lede #7343: AX6，5Ghz，160M，NSS启动报错

Ricky-Hao, 2021-07-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：自己编译的AX6固件，纯净版本基本没有手动勾选额外包。在5G开启160M时，系统日志显示 `wlan0: Failed to create a NSS virtual interface`，而降到80M，NSS正常。

### (2) 路由器型号和固件版本

A：路由器：RedMi AX6，固件版本：`OpenWrt R21.7.1.100 / LuCI Master (git-21.188.53404-317d4ad)`

### (3) 详细日志

A：
```
Sun Jul 11 10:54:02 2021 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sun Jul 11 10:54:02 2021 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Sun Jul 11 10:54:02 2021 kern.err kernel: [36873.077102] wlan0: Failed to create a NSS virtual interface
Sun Jul 11 10:54:02 2021 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sun Jul 11 10:54:02 2021 kern.info kernel: [36873.123947] br-lan: port 4(wlan0) entered blocking state
Sun Jul 11 10:54:02 2021 kern.info kernel: [36873.123986] br-lan: port 4(wlan0) entered disabled state
Sun Jul 11 10:54:02 2021 kern.info kernel: [36873.129335] device wlan0 entered promiscuous mode
Sun Jul 11 10:54:02 2021 kern.info kernel: [36873.133865] br-lan: port 4(wlan0) entered blocking state
Sun Jul 11 10:54:02 2021 kern.info kernel: [36873.138434] br-lan: port 4(wlan0) entered forwarding state
Sun Jul 11 10:54:02 2021 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Jul 11 10:54:02 2021 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Sun Jul 11 10:54:03 2021 daemon.err hostapd: Interface initialization failed
Sun Jul 11 10:54:03 2021 daemon.notice hostapd: wlan0: interface state HT_SCAN->DISABLED
Sun Jul 11 10:54:03 2021 daemon.notice hostapd: wlan0: AP-DISABLED
Sun Jul 11 10:54:03 2021 kern.info kernel: [36873.741043] br-lan: port 4(wlan0) entered disabled state
```

kisaragychihaya, 2021-07-14:
同这个，还有wlan0: Failed to create a NSS virtual interface 这个错误 关闭160MHz还有

Mitsuhaxy, 2021-07-16:
请问开启160Mhz的情况下能搜到WiFi信号吗，我一直都搜不到

bilibilistack, 2021-09-02:
我的AX6会概率出现断流，重新连接显示正在获取IP地址。80mhz也这样

XiaoliChan, 2021-10-12:
> 我的AX6会概率出现断流，重新连接显示正在获取IP地址。80mhz也这样

What mode? dumb ap or net gateway?

closed 2023-12-22 by aiamadeus