coolsnowwolf/lede #7355: 中继模式下IPV6无网络

slipdaly, 2021-07-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：中继模式下IPV6无网络
op做二级路由，配置ipv6后，电脑等设备可以获取到op分配的ipv6地址但是ipv6没网络
[image]

[image]
wwan是连接的上级无线
wwan6是新建的v6接口
[image]
[image]
路由通告服务、DHCPv6 服务、总是通告默认路由，这里的选项各种组合都尝试了，比如改成中继模式，还是不行。

最终是通过添加脚本解决的，是我哪里配置不对吗？编译时已经选择了ipv6helper
```shell
#!/bin/sh
# filename: /etc/hotplug.d/iface/90-ipv6
# please make sure this file has permission 644

# check if it is the intf which has a public ipv6 address like "2001:da8:100d:aaaa:485c::1/64"
interface_public="wwan6"
[ "$INTERFACE" = "$interface_public" ] || exit 0

# Set "IPv6 ULA Prefix"
uci set network.globals.ula_prefix="fd00:1:8::/64"
uci commit network

# Set the DHCP server to "Always announce default router"
uci set dhcp.lan.dhcpv6=server
uci set dhcp.lan.ra=server
uci set dhcp.lan.ra_default=1
uci set dhcp.lan.ra_management=1
uci commit dhcp

res=`ip -6 route | grep "default from"`
gateway=`echo $res | awk '{print $5}'`
interface=`echo $res | awk '{print $7}'`

if [ "$ACTION" = ifup ]; then
    ip -6 r add default via $gateway dev $interface
    if !(ip6tables-save -t nat | grep -q "v6NAT"); then
        ip6tables -t nat -A POSTROUTING -o $interface -m comment --comment "v6NAT" -j MASQUERADE
    fi
else
    ip6tables -t nat -D POSTROUTING -o $interface -m comment --comment "v6NAT" -j MASQUERADE
    ip -6 r del default via $gateway dev $interface
fi
```

### (2) 路由器型号和固件版本

A：
主机型号 Phicomm PSG1218
固件版本 OpenWrt R21.7.1.100 / LuCI Master (git-21.188.53404-317d4ad)
内核版本 5.4.128

### (3) 详细日志

A：需要哪些日志，我收集下

huaihuai0406, 2021-07-15:
NDP一般配置 中继或者混合模式，dns配置好 终端获取到240开头的电信ipv6地址即可

slipdaly, 2021-07-15:
> NDP一般配置 中继或者混合模式，dns配置好 终端获取到240开头的电信ipv6地址即可

各种组合配置都试过了，没用。op做一级路由是没问题的。只有做二级路由才有问题。

huaihuai0406, 2021-07-16:
> > NDP一般配置 中继或者混合模式，dns配置好 终端获取到240开头的电信ipv6地址即可
> 
> 各种组合配置都试过了，没用。op做一级路由是没问题的。只有做二级路由才有问题。

做二级路由 一级路由也需要按照上边的开启中继才可以

slipdaly, 2021-07-19:
> > > NDP一般配置 中继或者混合模式，dns配置好 终端获取到240开头的电信ipv6地址即可
> > 
> > 
> > 各种组合配置都试过了，没用。op做一级路由是没问题的。只有做二级路由才有问题。
> 
> 做二级路由 一级路由也需要按照上边的开启中继才可以

一级路由不是op呀，是辣鸡TP-XDR3020

closed 2024-11-19 by coolsnowwolf