coolsnowwolf/lede #7358: 最新内核的adblock的后进不去设置界面

liuyanantj, 2021-07-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：自己喜欢用adblock，但是最新版本内核编译后，进入adblock不能显示设置页面 只显示一串代码 其他插件正常 显示的代码见下方

### (2) 路由器型号和固件版本

A：x86 j4125 软路由

### (3) 详细日志

/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute firstchild dispatcher target for entry '/admin/services/adblock'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/adblock/tab_from_cbi'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/adblock/overview_tab.lua:30: attempt to index field 'data' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

Sinmists, 2021-07-23:
相同问题，日志如下 

/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute firstchild dispatcher target for entry '/admin/services/adblock'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/adblock/tab_from_cbi'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/adblock/overview_tab.lua:21: attempt to index field 'data' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

PanXinQiang, 2021-08-18:
相同问题

/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute firstchild dispatcher target for entry '/admin/services/adblock'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/adblock/tab_from_cbi'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/adblock/overview_tab.lua:30: attempt to index field 'data' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

alllexx88, 2021-10-01:
The same problem still holds

```
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute firstchild dispatcher target for entry '/admin/services/adblock'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/adblock/tab_from_cbi'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/adblock/overview_tab.lua:30: attempt to index field 'data' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
```

closed 2021-10-07 by coolsnowwolf