coolsnowwolf/lede #7359: 第一次编译出错

ico228, 2021-07-12:
make[3]: Leaving directory '/home/st/lede/feeds/packages/net/wget'
time: package/feeds/packages/wget/ssl/compile#0.21#0.03#0.22
make[3]: Entering directory '/home/st/lede/feeds/kenzo/luci-app-adguardhome'
rm -rf /home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome.installed /home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome
mkdir -p /home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome
install -d -m0755 /home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome/usr/lib/lua/luci
cp -pR ./luasrc/* /home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome/usr/lib/lua/luci
install -d -m0755 /home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome/
cp -pR ./root/* /home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome/
install -d -m0755 /home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome/usr/lib/lua/luci/i18n
po2lmo ./po/zh-cn/AdGuardHome.po /home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome/usr/lib/lua/luci/i18n/AdGuardHome.zh-cn.lmo
bash: po2lmo: command not found
Makefile:72: recipe for target '/home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome.installed' failed
make[3]: *** [/home/st/lede/build_dir/target-x86_64_musl/luci-app-adguardhome/.pkgdir/luci-app-adguardhome.installed] Error 127
make[3]: Leaving directory '/home/st/lede/feeds/kenzo/luci-app-adguardhome'
time: package/feeds/kenzo/luci-app-adguardhome/compile#0.15#0.04#0.18
    ERROR: package/feeds/kenzo/luci-app-adguardhome failed to build.
package/Makefile:114: recipe for target 'package/feeds/kenzo/luci-app-adguardhome/compile' failed
make[2]: *** [package/feeds/kenzo/luci-app-adguardhome/compile] Error 1
make[2]: Leaving directory '/home/st/lede'
package/Makefile:110: recipe for target '/home/st/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/st/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/st/lede'
/home/st/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

minlang112, 2021-07-12:
bash: po2lmo: command not found

你可能需要先安装这个东西 po2lmo

wggglggg, 2021-07-15:
adguardhome  钩去掉试试

closed 2022-02-12 by Beginner-Go