coolsnowwolf/lede #7369: R21.7.1.100，Privoxy的luci界面报500错误

NagaseKouichi, 2021-07-13:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R21.7.1.100，Privoxy的luci界面报500错误，5.4内核的版本中没有这个问题

### (2) 路由器型号和固件版本

A：X86软路由，R21.7.1.100

### (3) 详细日志

[image]

1757750341, 2021-07-15:
跟你一样，最近好几个版本都报错500，手头设备有斐讯K2(mt7620)、小米mini(mt7620)、K2P(mt7621)、360P2(mt7628)，都一样报错，以前的老版本就不报错，但是git pull后就回不去了
[image]
[image]

NagaseKouichi, 2021-07-24:
> 跟你一样，最近好几个版本都报错500，手头设备有斐讯K2(mt7620)、小米mini(mt7620)、K2P(mt7621)、360P2(mt7628)，都一样报错，以前的老版本就不报错，但是git pull后就回不去了
> [image]
> [image]

在L大的帮助下解决了，已经提交更新，更新下feed就可以了。

closed 2021-07-25 by NagaseKouichi