coolsnowwolf/lede #7371: HC5962无法升级最新编译的固件

rshi0212, 2021-07-13:
显示 “不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件。"

同时编译的CR660X可以升级。

[image]
[image]
[image]

rshi0212, 2021-07-13:
CR660X可以升级~
[image]

weweyes, 2021-07-13:
小米路由器AC2100也是无法升级

weweyes, 2021-07-13:
> 对啊，因为之前是交换机架构的，这次统一换dsa了

大佬，请问一下，如果想升级的话，是不是得从breed控制台升级

rshi0212, 2021-07-13:
是不是需要从breed用factory.bin刷机升级？

vsvkhs, 2021-07-17:
AC85P同样的情况，无法从5.4.128升级到5.4.132

dukmonki, 2021-07-24:
编译时，先不要选择太多软件，小于16M，可编译产生factory 固件。在BREED状态下刷此固件，再刷升级固件。

vsvkhs, 2021-08-11:
请问怎么弄呀，5.4.125升级5.4.138和你图片显示的一样“不支持所上传的映像文件格式.请选择适合当前平台的通用映像文件.”

EnnawYang, 2021-08-11:
breed，please

vsvkhs, 2021-08-11:
> 品种，请

> breed，please

AC85P not have breed

vsvkhs, 2021-08-11:
> breed，please

AC85P not have breed

vsvkhs, 2021-08-11:
解决了，亲测强制刷机不保留配置可行。“sysupgrade”是你的固件名（随便改但要一致）用winscp把固件上传到tmp目录下，然后用命令刷机等会自动重启：sysupgrade -F -n /tmp/sysupgrade.bin

closed 2021-09-14 by AmadeusGhost