coolsnowwolf/lede #7375: DIR-882-A1固件无法启动路由器

blueveryday, 2021-07-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新的源代码，编译DIR-882-A1固件，编译没有报错。breed里面刷了以后，路由器无法开机，左边1个等一直狂闪。

### (2) 路由器型号和固件版本

A：DLINK DIR-882-A1

### (3) 详细日志

A：

AmadeusGhost, 2021-07-14:
有办法提供 ttl log 么，我昨天已经改回 breed 分区了（

blueveryday, 2021-07-14:
> 有办法提供 ttl log 么，我昨天已经改回 breed 分区了（

总裁，在immortalwrt的2021年5月29日的commit里面，编译是可以正常启动路由器的。
git reset --hard ec4852f7313cc377785ce27cac51a71890ab20d6
后来有个德国人提交了factory的修改，好像就有问题了。
我也是帮外地一个朋友在弄，机子不在我身边，ttl log有点困难。

blueveryday, 2021-07-14:
> 有办法提供 ttl log 么，我昨天已经改回 breed 分区了（

在dir-882-a1的breed里面有分区的多个选项，不知道选哪个正确，默认选的公版分区，然后刷的。

blueveryday, 2021-07-14:
> 有办法提供 ttl log 么，我昨天已经改回 breed 分区了（

这是immortalwrt的mt7621_dlink_flash-16m-a1.dtsi文件

>
’ // SPDX-License-Identifier: GPL-2.0-or-later OR MIT

&spi0 {
	status = "okay";

	flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <50000000>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "u-boot";
				reg = <0x0 0x30000>;
				read-only;
			};

			partition@30000 {
				label = "u-boot-env";
				reg = <0x30000 0x10000>;
				read-only;
			};

			factory: partition@40000 {
				label = "factory";
				reg = <0x40000 0x20000>;
				read-only;
			};

			partition@60000 {
				compatible = "openwrt,uimage", "denx,uimage";
				openwrt,padding = <96>;
				label = "firmware";
				reg = <0x60000 0xfa0000>;
			};
		};
	};
};
‘

AmadeusGhost, 2021-07-14:
> 这是immortalwrt的mt7621_dlink_flash-16m-a1.dtsi文件

这个 dtsi 肯定不行的，写的是原厂分区，从 breed 刷入会找不到内核

closed 2021-07-14 by blueveryday