coolsnowwolf/lede #7376: 端口转发部分有效

wangweitung, 2021-07-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用端口转发，内网的ubuntu server开放1001端口，内网/外网ip+1001可以访问。
                                内网的群晖开放5000端口，内网ip+5000可以访问，外网ip+5000无法访问。

### (2) 路由器型号和固件版本

A： 路由器型号：Intel(R) Celeron(R) 3205U @ 1.50GHz : 2 Core 2 Thread (CpuMark : 21608.773162 Scores) 
       固件版本：OpenWrt R21.7.1.100 / LuCI Master (git-21.188.53404-317d4ad) 
       内核版本：5.10.46

### (3) 详细日志

A：自己本地编译，固件，仅编译了vssr等常用固件。无turbcc。
B：安装方法：全新安装，全新配置。未使用原来的配置。
C：在防火墙自定义规则中加入以下规则，也无法访问：
`iptables -t nat -I PREROUTING -p tcp --dport 5000 -j DNAT --to 192.168.1.188`
`iptables -t nat -I PREROUTING -p udp --dport 5000 -j DNAT --to 192.168.1.188`

wangweitung, 2021-07-14:
补充一下，我也没有编译docker

wangweitung, 2021-07-14:
自我排查了下，可能有两个原因。
1.群晖设置有问题。
2.路由器5000端口被占用？

wangweitung, 2021-07-15:
1.换了个源码编译
2.强制重启了群晖。
可以转发了，不需要任何多余的设置。
issue closed。

closed 2021-07-15 by wangweitung

lovelyOK, 2021-07-19:
清问，你换个源码，是OPENWRT换了一个，还是换了一个LEDE日期版本？我也遇到这样的问题，早期6月之前的LEDE版本还不存在这个现象。

> 1.换了个源码编译
> 2.强制重启了群晖。
> 可以转发了，不需要任何多余的设置。
> issue closed。

wangweitung, 2021-07-19:
> 
> 
> 清问，你换个源码，是OPENWRT换了一个，还是换了一个LEDE日期版本？我也遇到这样的问题，早期6月之前的LEDE版本还不存在这个现象。
> 
> > 1.换了个源码编译
> > 2.强制重启了群晖。
> > 可以转发了，不需要任何多余的设置。
> > issue closed。

1、编译了：https://github.com/Lienol/openwrt，发现没有端口转发失效的问题。
2、切回https://github.com/coolsnowwolf/lede，重新拉取源码，编译。
3、重启群晖。
4、OK了。

综上，个人觉得是群晖的问题，lede固件无问题。