coolsnowwolf/lede #7379: 红米ac2100怎么选闭源驱动

Ozzy-hzy, 2021-07-14:
我把，kmod里的全部取消勾选，然后选上luci-app-mtwifi，编译出来无线开不了。
开源驱动很不稳定，下载只能跑几十兆，上传看着没问题。
[image]

Neverends4, 2021-07-16:
默认就是闭源驱动啊？kmod是内核模块，你不选如何驱动无线？！
最简单的是config删掉重新选一次2100机型，配置就回来了。
另外我红米2100用开源也没发现什么问题，当然无线只看视频不玩游戏。

Ozzy-hzy, 2021-07-16:
> 默认就是闭源驱动啊？kmod是内核模块，你不选如何驱动无线？！
> 最简单的是config删掉重新选一次2100机型，配置就回来了。
> 另外我红米2100用开源也没发现什么问题，当然无线只看视频不玩游戏。

默认好像是开源的，信号很差，我选了kmod-7603e和kmod-7615d和luci-app-mtwifi编译出来没有2.4g信号

Neverends4, 2021-07-17:
难道更新后默认开源驱动了？不是说默认7615用闭源支持HWNAT么？
我很久没刷机了，稳定就够，所以不清楚什么情况。
如果你要切，有两步，
1.取消开源驱动：kmod-mt7603、kmod-mt7615-fireware和 kmod-mt7615e，以及wpad-openssl （network-wirelessAPD下面，可能是wpad-mini或其他的，总之把wpad开头的都取消否则会冲突）
2.选择闭源驱动：mtwifi、kmod-mt7603e、kmod-mt7615d
如果编译失败清空一下再来

Ozzy-hzy, 2021-07-19:
> 难道更新后默认开源驱动了？不是说默认7615用闭源支持HWNAT么？
> 我很久没刷机了，稳定就够，所以不清楚什么情况。
> 如果你要切，有两步，
> 1.取消开源驱动：kmod-mt7603、kmod-mt7615-fireware和 kmod-mt7615e，以及wpad-openssl （network-wirelessAPD下面，可能是wpad-mini或其他的，总之把wpad开头的都取消否则会冲突）
> 2.选择闭源驱动：mtwifi、kmod-mt7603e、kmod-mt7615d
> 如果编译失败清空一下再来

按着你的又编译一次，出来2.4wifi还是不能开启，5g正常，如图
[image]
点启用就提示那个
这是选的驱动，那个wpad-openssl也取消了
[image]

Neverends4, 2021-07-19:
那就不知道了，估计是最近大升级后配置有变化
我在4月整过闭源驱动，就是用上面的配置，2.4、5G无线都正常。
后面因为HWNAT导致频繁重启，不用HWNAT就直接切开源驱动了。
开源驱动不支持HWNAT，5g可以到480M，我不玩游戏足够了，就没有再用闭源。

Ozzy-hzy, 2021-07-19:
> 那就不知道了，估计是最近大升级后配置有变化
> 我在4月整过闭源驱动，就是用上面的配置，2.4、5G无线都正常。
> 后面因为HWNAT导致频繁重启，不用HWNAT就直接切开源驱动了。
> 开源驱动不支持HWNAT，5g可以到480M，我不玩游戏足够了，就没有再用闭源。

现在ssrplu开了还很容易爆内存，直接变成未启用。不折腾了，我也刷回去年编译的固件了。。。

Henshufu, 2021-07-19:
我按三楼编译的，2.4G和5G都正常。

Ozzy-hzy, 2021-07-19:
> 我按三楼编译的，2.4G和5G都正常。

啊？是最新的吗

Neverends4, 2021-07-21:
不对吧，我特意用最新源码git-21.200.60006-aa12cbf编译，内核和你对不上啊，我这边内核是5.4.125，mt-wifi版本倒是一样
目前2.4/5G一切正常。
[image]

话说最新版无线还是没有HWNAT，，，

Ozzy-hzy, 2021-07-21:
> 不对吧，我特意用最新源码R21.6.14 编译，内核和你对不上啊，我这边内核是5.4.125，mt-wifi版本倒是一样
> 目前2.4/5G一切正常。
> [image]
> 
> 话说最新版无线还是没有HWNAT，，，

最新源码不是R21.7.15吗？

Neverends4, 2021-07-22:
忘记pull了……不过拉完源码，2100双无线还是都正常啊
[image]
另外大更新后貌似HWNAT效果减弱了，以前有线可以跑700M现在只有500M，而且htop看CPU占用有soft-irq（紫色），大更新以前是完全没有CPU占用显示的，现在好像和官方op的MT6721情况类似了
闭源驱动无线5g现在测速只有280M………………我觉得还不如开源驱动……

Neverends4, 2021-07-22:
看起来好像是之前遗留源码导致我的内核还停在5.4，我试试清空重新编译。

Ozzy-hzy, 2021-07-22:
> 看起来好像是之前遗留源码导致我的内核还停在5.4，我试试清空重新编译。

麻烦了，还有ssrplus TCP和UDP全开的话容易爆内存，编译的时候可以怎么优化下吗

Neverends4, 2021-07-22:
奇怪，文件夹删了整个重新编译+配置，还是5.4，貌似内核就是5.4.132了
反正这个内核无线正常………………
ssr优化要问作者了。然而这机子只有128M内存，再优化就这个样，定位应该就是AP+轻量任务，不是主路由，有其他需求还是要用软路由。

Henshufu, 2021-07-22:
奇怪之前的回复怎么没了，我是7月17日晚云编译的。
<html>
<body>

固件版本 | OpenWrt R21.7.15 / LuCI Master (git-21.197.21199-cd0ddb8)
-- | --
内核版本 | 5.10.50

</body>
</html>

Ozzy-hzy, 2021-07-22:
> 奇怪之前的回复怎么没了，我是7月17日晚云编译的。
> 
> 固件版本	OpenWrt R21.7.15 / LuCI Master (git-21.197.21199-cd0ddb8)
> 内核版本	5.10.50

那就奇怪了，我可能要再make clean一下重新编译一次

Ozzy-hzy, 2021-07-23:
今天make clean后重新编译了一遍，版本R21.7.15，内核5.4.132，闭源wifi正常了
[image]

closed 2021-07-23 by Ozzy-hzy

neowong, 2022-05-01:
为啥我不管用开源还是闭源都在界面中无显示，看.config中luci-app-mtwifi已经选中，make menuconfig中没有该选项