coolsnowwolf/lede #7383: （默认配置）第一次编译报错 Ubuntu

1-bytes, 2021-07-14:
### **系统信息：**
香港服务器
LSB Version:    core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal

### **错误信息：**
`SHELL= flock /home/enzo/lede/tmp/.sysfsutils-2.1.0.tar.gz.flock -c '    /home/enzo/lede/scripts/download.pl "/home/enzo/lede/dl" "sys                                                                         fsutils-2.1.0.tar.gz" "e865de2c1f559fff0d3fc936e660c0efaf7afe662064f2fb97ccad1ec28d208a" "" "@SF/linux-diag"    '
mkdir -p /home/enzo/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '2' | cmp -s - /home/enzo/lede/staging_dir/target-x86_64_musl/pkginfo/libsysfs.version || { echo '2' > /home/enzo/lede/staging_d                                                                         ir/target-x86_64_musl/pkginfo/libsysfs.version;  }
SHELL= flock /home/enzo/lede/tmp/.root-copy.flock -c 'cp -fpR /home/enzo/lede/build_dir/target-x86_64_musl/sysfsutils-2.1.0/.pkgdir/l                                                                         ibsysfs/. /home/enzo/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/enzo/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libsysfs_installed
echo "libsysfs" >> /home/enzo/lede/staging_dir/target-x86_64_musl/pkginfo/sysfsutils.default.install
mkdir -p /home/enzo/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/enzo/lede/tmp/.root-copy.flock -c 'cp -fpR /home/enzo/lede/build_dir/target-x86_64_musl/sysfsutils-2.1.0/.pkgdir/s                                                                         ysfsutils/. /home/enzo/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/enzo/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.sysfsutils_installed
echo "sysfsutils" >> /home/enzo/lede/staging_dir/target-x86_64_musl/pkginfo/sysfsutils.default.install
make[3]: Leaving directory '/home/enzo/lede/package/libs/sysfsutils'
time: package/libs/sysfsutils/compile#0.16#0.01#0.16
make[3]: Entering directory '/home/enzo/lede/feeds/packages/utils/lm-sensors'
mkdir -p /home/enzo/lede/dl
SHELL= flock /home/enzo/lede/tmp/.lm-sensors-3.6.0.tar.gz.flock -c '    /home/enzo/lede/scripts/download.pl "/home/enzo/lede/dl" "lm-                                                                         sensors-3.6.0.tar.gz" "0591f9fa0339f0d15e75326d0365871c2d4e2ed8aa1ff759b3a55d3734b7d197" "" "https://codeload.github.com/lm-sensors/l                                                                         m-sensors/tar.gz/V3-6-0?"    '
rm -f /home/enzo/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.built
touch /home/enzo/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.built_check
make  -C /home/enzo/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0 AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux                                                                         -musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-                                                                         prefix-map=/home/enzo/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0=lm-sensors-3-6-0 -Wformat -Werror=format-security -fstack-pr                                                                         otector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64                                                                         -openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gc                                                                         c-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump S                                                                         IZE=x86_64-openwrt-linux-musl-size KERNELVERSION="" LINUX="" CC="x86_64-openwrt-linux-musl-gcc" CFLAGS="-Os -pipe -fno-caller-saves -                                                                         fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/enzo/lede/build_dir/targe                                                                         t-x86_64_musl/lm-sensors-3-6-0=lm-sensors-3-6-0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl                                                                         ,-z,relro" CPPFLAGS="-I/home/enzo/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/enzo/lede/staging_dir/toolchai                                                                         n-x86_64_gcc-8.4.0_musl/include/fortify -I/home/enzo/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" STAGING_DIR="/home/enz                                                                         o/lede/staging_dir/target-x86_64_musl" PREFIX="/usr" MACHINE="x86_64" EXLDFLAGS="" user
make[4]: Entering directory '/home/enzo/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0'
x86_64-openwrt-linux-musl-gcc  -o prog/sensors/sensors prog/sensors/main.ro prog/sensors/chips.ro -liconv -Llib -lsensors
/home/enzo/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-                                                                         musl/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status
make[4]: *** [prog/sensors/Module.mk:43: prog/sensors/sensors] Error 1
make[4]: Leaving directory '/home/enzo/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0'
make[3]: *** [Makefile:129: /home/enzo/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.built] Error 2
make[3]: Leaving directory '/home/enzo/lede/feeds/packages/utils/lm-sensors'
time: package/feeds/packages/lm-sensors/compile#0.13#0.02#0.14
    ERROR: package/feeds/packages/lm-sensors failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/lm-sensors/compile] Error 1
make[2]: Leaving directory '/home/enzo/lede'
make[1]: *** [package/Makefile:110: /home/enzo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/enzo/lede'
make: *** [/home/enzo/lede/include/toplevel.mk:230：world] 错误 2
`

**### 个人已经处理过的步骤：**
看到是lm-sensors包编译的时候报错了：cannot find -liconv  推断是没有找到libiconv这个包导致的
我去系统里面检查libiconv是存在的
如图：
[image]
于是猜想是libiconv包的版本不对，根据 https://www.gnu.org/software/libiconv/ 的指引去编译安装了 1.16的包

随后清理掉之前下载和编译的程序：
$ make dirclean
$ make distclean
$ git clean -xdf

清理完重新一步一步的进行编译步骤，报错依旧 :(

wggglggg, 2021-07-14:
是的, 无法编译...报错.

wggglggg, 2021-07-14:
a - bn_s_mp_prime_is_divisible.o
a - bn_s_mp_rand_jenkins.o
a - bn_s_mp_rand_platform.o
a - bn_s_mp_reverse.o
a - bn_s_mp_sqr.o
a - bn_s_mp_sqr_fast.o
a - bn_s_mp_sub.o
a - bn_s_mp_toom_mul.o
a - bn_s_mp_toom_sqr.o
x86_64-openwrt-linux-musl-gcc-ranlib libtommath.a
make[5]: Leaving directory '/home/wggglggg/lede/build_dir/target-x86_64_musl/dropbear-2020.81/libtommath'
x86_64-openwrt-linux-musl-gcc -L/home/wggglggg/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/wggglggg/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -DPIC -fpic -specs=/home/wggglggg/lede/include/hardened-ld-pie.specs -znow -zrelro -Wl,--gc-sections -flto=jobserver  -o dropbearmulti dbmulti.o atomicio.o bignum.o buffer.o chachapoly.o circbuffer.o cli-agentfwd.o cli-auth.o cli-authinteract.o cli-authpasswd.o cli-authpubkey.o cli-channel.o cli-chansession.o cli-kex.o cli-main.o cli-runopts.o cli-session.o cli-tcpfwd.o common-algo.o common-channel.o common-chansession.o common-kex.o common-runopts.o common-session.o compat.o crypto_desc.o curve25519.o dbhelpers.o dbmalloc.o dbrandom.o dbutil.o dh_groups.o dropbearkey.o dss.o ecc.o ecdsa.o ed25519.o fake-rfc2553.o gcm.o gendss.o gened25519.o genrsa.o gensignkey.o list.o listener.o loginrec.o ltc_prng.o netio.o packet.o process-packet.o progressmeter.o queue.o rsa.o scp.o scpmisc.o signkey.o sshpty.o svr-agentfwd.o svr-auth.o svr-authpam.o svr-authpasswd.o svr-authpubkey.o svr-authpubkeyoptions.o svr-chansession.o svr-kex.o svr-main.o svr-runopts.o svr-service.o svr-session.o svr-tcpfwd.o svr-x11fwd.o tcp-accept.o termcodes.o libtomcrypt/libtomcrypt.a libtommath/libtommath.a  -lcrypt
make[5]: *** [/home/wggglggg/lede/tmp/ccLyV7Ck.ltrans1.ltrans.o] Error 1
lto-wrapper: fatal error: make returned 2 exit status
compilation terminated.
/home/wggglggg/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
Makefile:222: recipe for target 'dropbearmulti' failed
make[4]: *** [dropbearmulti] Error 1
make[4]: Leaving directory '/home/wggglggg/lede/build_dir/target-x86_64_musl/dropbear-2020.81'
Makefile:213: recipe for target '/home/wggglggg/lede/build_dir/target-x86_64_musl/dropbear-2020.81/.built' failed
make[3]: *** [/home/wggglggg/lede/build_dir/target-x86_64_musl/dropbear-2020.81/.built] Error 2
make[3]: Leaving directory '/home/wggglggg/lede/package/network/services/dropbear'
time: package/network/services/dropbear/compile#60.71#36.17#89.74
    ERROR: package/network/services/dropbear failed to build.
package/Makefile:114: recipe for target 'package/network/services/dropbear/compile' failed
make[2]: *** [package/network/services/dropbear/compile] Error 1
make[2]: Leaving directory '/home/wggglggg/lede'
package/Makefile:110: recipe for target '/home/wggglggg/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/wggglggg/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wggglggg/lede'
/home/wggglggg/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

1-bytes, 2021-07-14:
我晚上重新换CentOS7.9进行尝试
猜测可能是系统问题，找个干净的系统试试

wggglggg, 2021-07-14:
我用的本地Ubuntu 18,五月份还能编译的，之前我编过很多次了，今天pull 了之后，又update install feeds后，就不行了，其它都没动过

发自我的iPhone

1-bytes, 2021-07-14:
> error: lto-wrapper failed

error: lto-wrapper failed

可能是项目新增或升级了某些包导致的，你可以执行 make dirclean 以后再编译尝试。
我看到你的错误和我的还不一样。

wggglggg, 2021-07-14:
我明天换别的试试

发自我的iPhone

wggglggg, 2021-07-15:
我把lede文件夹删了，重clone lede.结果编译也出错，于是把Ubuntu 重装，编译也出错，后来menuconfig时，我把luck 里默钩选的工具去掉钩，只钩选password ssr 重启 关机 ，这次编译成了。

发自我的iPhone

1-bytes, 2021-07-15:
我原本的系统环境可能有问题才导致的各种报错，后来重新搞了一台服务器专门用于编译，同样是Ubuntu20系统，已经编译成功了，稍等下把固件刷进去

这个issues先关闭了，可以加我wx 互相沟通解决
base64解密就是我wx：MTk5MTk5NjU0Mw==

closed 2021-07-15 by 1-bytes