coolsnowwolf/lede #7384: 跑docker下应用流量不走ssr

wxs-dev, 2021-07-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
设备是N1，做主路由拨号上网
局域网网段192.168.4.0/24
docker是桥接，网段171.17.0.0/16
开ssr plus+后，局域网的手机电脑可以科学上网。但跑docker下应用不能科学上网，可以访问国内的网站
请问这是什么问题？

wtbam, 2021-07-16:
在ssr里设置强制走代理

wxs-dev, 2021-07-16:
有没有朋友可以帮忙回答，困扰了好几天，实在没辙了😭

acooler15, 2021-07-16:
我用的 [dockerman](https://github.com/lisaac/luci-app-dockerman)+ dockerd来管理docker，在ssr的`访问控制-接口控制` 中有一个`docker`接口，勾选上就可以访问了

wxs-dev, 2021-07-16:
> 我用的 [dockerman](https://github.com/lisaac/luci-app-dockerman)+ dockerd来管理docker，在ssr的`访问控制-接口控制` 中有一个`docker`接口，勾选上就可以访问了（docker容器的网络模式是默认的bridge，其它的没试过）
 
[image]
[image]

我这没有docker接口，好像也没用dockerman

acooler15, 2021-07-17:
可以试试dockerman呀，感觉挺好用的。。。

643406834, 2021-07-29:
environment:
    - http_proxy=http://10.0.0.1:7880
    - HTTP_PROXY=http://10.0.0.1:7880
    - https_proxy=http://10.0.0.1:7880
    - HTTPS_PROXY=http://10.0.0.1:7880
docker应用的环境变量全部指向ssr

wxs-dev, 2021-08-14:
> environment:
> - http_proxy=http://10.0.0.1:7880
> - HTTP_PROXY=http://10.0.0.1:7880
> - https_proxy=http://10.0.0.1:7880
> - HTTPS_PROXY=http://10.0.0.1:7880
> docker应用的环境变量全部指向ssr

ssr+只能socket5代理
另外还有其他的办法吗，这样的话，每个容器都需要配置这个环境

acooler15, 2021-08-16:
> > environment:
> > 
> > * http_proxy=http://10.0.0.1:7880
> > * HTTP_PROXY=http://10.0.0.1:7880
> > * https_proxy=http://10.0.0.1:7880
> > * HTTPS_PROXY=http://10.0.0.1:7880
> >   docker应用的环境变量全部指向ssr
> 
> ssr+只能socket5代理
> 另外还有其他的办法吗，这样的话，每个容器都需要配置这个环境

不需要一个个的设置，可以在`~/.docker/config.json`中设置`httpProxy`、`httpsProxy`，然后再重新创建容器。https://docs.docker.com/engine/reference/commandline/cli/#change-the-docker-directory

acooler15, 2021-08-16:
> > 我用的 [dockerman](https://github.com/lisaac/luci-app-dockerman)+ dockerd来管理docker，在ssr的`访问控制-接口控制` 中有一个`docker`接口，勾选上就可以访问了（docker容器的网络模式是默认的bridge，其它的没试过）
> 
> [image]
> [image]
> 
> 我这没有docker接口，好像也没用dockerman

这是我的network配置文件：
```
config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd78:4736:6d25::/48'

config interface 'lan'
        option type 'bridge'
        option ifname 'eth0'
        option proto 'static'
        option ipaddr '192.168.123.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'docker'
        option ifname 'docker0'
        option proto 'none'
        option auto '0'

config device 'docker0'
        option type 'bridge'
        option name 'docker0'
        list ifname 'docker0'

config interface 'wan'
        option proto 'pppoe'
        option ifname 'eth1'
        option username '0'
        option password '0'
        option ipv6 'auto'
        option peerdns '0'
        option dns '223.5.5.5 223.6.6.6 2400:3200::1 2400:3200:baba::1'
        option keepalive '0'
        option mtu '1480'
```

klaus121, 2021-11-17:
> 有没有朋友可以帮忙回答，困扰了好几天，实在没辙了😭

解决了吗，换ssr版本吗？

Xunflash, 2022-01-27:
> 我用的 [dockerman](https://github.com/lisaac/luci-app-dockerman)+ dockerd来管理docker，在ssr的`访问控制-接口控制` 中有一个`docker`接口，勾选上就可以访问了（docker容器的网络模式是默认的bridge，其它的没试过）

有个问题，我需要固定IP，因此自己新建了一个bridge网络，然而这个网络没有父接口，导致ssr勾选上docker0接口无效...
而且我的$HOME目录下面也没有.docker目录..

closed 2024-11-19 by coolsnowwolf