coolsnowwolf/lede #7389: 编译wrt1200ac的问题

rshi0212, 2021-07-14:
请问编译wrt1200ac时提示的Architected Timer Support是什么选项，应该选YES还是NO？
[image]

1-bytes, 2021-07-14:
目前遇到了和楼主一样的问题... uugamebooster这个包失效了 ，下载是404页面，编译直接不通过

HoldOnBro, 2021-07-14:
boom

adamgonglj, 2021-07-15:
同样的问题。

LOVEChen, 2021-07-15:
SHELL= flock /root/lede/tmp/.uugamebooster-x86_64-v2.12.0.tar.gz.flock -c '     /root/lede/scripts/download.pl "/root/lede/dl" "uugamebooster-x86_64-v2.12.0.tar.gz" "61fffa3e17f21cca78c85b0f0bbcd7ed" "" "http://uu.gdl.netease.com/openwrt-x86_64/v2.12.0/uu.tar.gz?"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://uu.gdl.netease.com/openwrt-x86_64/v2.12.0/uu.tar.gz?/uugamebooster-x86_64-v2.12.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found

haig233, 2021-07-15:
同样，uugame无法下载

NagaseKouichi, 2021-07-15:
uugame最新版不是2.10.0吗？下载地址里面怎么是2.12.0

closed 2021-07-15 by coolsnowwolf