coolsnowwolf/lede #7393: 关于 SSR Plus+ 的 dns 解析分流规则与 ip 代理分流规则的不一致问题分析与建议

lynncui00, 2021-07-15:
#### 本人情况
刚使用 lede 不到 2 天的纯小白一枚。

#### 直观体验存在的问题情况
- SSR Plus+ 使用 “绕大陆 IP 模式”，其他模式应该不存在这个问题现象。
- 开启 SSR Plus+ 后，访问某些国内网站的图片视频资源，明显比没开启的时候慢。

#### 问题简单分析
- 国内某些网站的 CDN 资源，解析出来的 IP 可能不一定在国内，也许在东南亚。
- 在代理的 ip 分流规则里面，这些 ip 会通过 vps 节点（美国）进行代理，导致链路物理距离变长，延迟变高。

#### SSR Plus+ dns 解析原理简单分析
- 域名解析的时候，GFW 名单里面的域名，SSR Plus+ 内部使用 vps 节点去解析，所以解析出来的结果一般离 vps 节点更近，这种域名大概率解析出来是国外 ip，所以最后通过 ip 分流走 vps 代理的话，理论上是最优的。
- 非 GFW 名单里面的域名，走国内 Dns 服务商 (默认应该是运营商给的) 解析，解析出来的结果离本人国内上网实际地点更近，这种域名大概率解析出来是国内 ip， 所以最后通过 ip 分流走直连，也是最优的。
- 问题出现在不一致的地方，就是非 GFW 名单的域名，解析出来是国外 ip， 同时最后 ip 分流走的是 vps 节点，这个在大部分情况下也许也是最优的。
- 唯一存在的问题就是前面描述的情况，全球 CDN 的资源解析出来在 东南亚，但最后绕行 美国 vps 节点。
- 所以本质上这个问题对于 SSR Plus+ 是一个很小的问题，但也不妨碍我们讨论这个问题。

#### 如何解决这种不一致问题？
- 无法通过域名判断解析出来的 ip 到底是国内还是国外的。
- 原则上只能通过统计和经验，通过域名名单的方式，无限逼近一致。
- 但域名解析千变万化，这本身不是一个靠谱的解决思路。

#### 必须从软件源头解决问题
- 由于目前通过远端 vps 解析 or 直连解析，是 SSR Plus+ 内部控制的逻辑，所以个人认为必须修改 SSR Plus+ 才能解决这个问题。
- 通过第三方的 dns 插件是无法规避这个问题的。

#### 解决思路建议
-  参考 google 的 [Backup Requests](http://static.googleusercontent.com/media/research.google.com/zh-CN//people/jeff/Berkeley-Latency-Mar2012.pdf)
- SSR Plus+ 对于所有域名（无论是否 GFW), 直接双发解析，R1. 通过 vps 节点走国外解析 R2. 直连国内解析
- 如果 R2 返回的是国内 ip，则直接使用 ip
- 如果 R2 返回的是国外 ip，则等待 R1 的结果，使用 R1 解析出来的 ip

#### 仍然存在的问题
- 如果 R2 解析出来的是国外 ip， 但 R1 解析出来的是国内 ip ？

_当然我在路由器这方面纯无知小白，接触的东西也不多，所以也许这个问题已经解决了？也是这些分析是错误的？_

EnnawYang, 2021-07-15:
1. 发错地方了
2. 能打这么多字肯定不是小白（

markvlenvision, 2021-07-15:
？（我是小白）

galaxyskyknight, 2021-07-15:
> ？（我是小白）

有感而发，有些人总是不会能好好的直白的讲人话告诉你应该发哪里，应该要如何如何，而是要绕着弯子的兜，或者只指出错误不详尽的讲解决方案，或者讲一半不讲一半，或者干脆不搭理。在于他们内心对小白或者一知半解或者对整个全貌了解的不够的user的comments或question的不屑和不耐。殊不知他们也是这样过来的。虽然道理上，没有谁有义务必须解答谁的问题，但是社区开源共享的灵魂，不就是应该互相帮助共享解决问题吗？可能在他们眼里他们只能和“高手”过招，回答“专业”问题吧，否则会掉了身价。我发现这样子的在打中国字的github里非常明显，在老外的github里，人家re的那真的是截然不同，不管你是一个dumb question还是一个professsional.（准备re“你也说了，没有谁有义务..../那你可以去老外那里”诸如此类云云的打住）谢谢。

fo2w, 2021-07-15:
楼上的老哥虽然有点暴躁, 但是说的是实情

hxlls, 2021-07-15:
1.这里不是55rp的项目地址，建议去到项目地址去讨论这个问题。
2.既然讨论了，我说下观点：55rp的解析机制我感觉是 在list 内的会走VPS 通过pdns去解析,不在list或者cidr 内的，走dnsmasq 去正常解析，在正常解析下，还有几种方式(新版的）缓存或者不缓存，所以我认为关键点是在list 的准确性，然而，list 准确性是靠另外一个源来搞定的，55rp 本身不负责这个东西,那个东西有独立的一个项目来负责，我觉得思路没问题，更多问题是列表内的东西准确性和完整性，这个是需要人去维护的，通过机器判定我觉得不现实。

lynncui00, 2021-07-15:
> 1.这里不是55rp的项目地址，建议去到项目地址去讨论这个问题。
> 2.既然讨论了，我说下观点：55rp的解析机制我感觉是 在list 内的会走VPS 通过pdns去解析,不在list或者cidr 内的，走dnsmasq 去正常解析，在正常解析下，还有几种方式(新版的）缓存或者不缓存，所以我认为关键点是在list 的准确性，然而，list 准确性是靠另外一个源来搞定的，55rp 本身不负责这个东西,那个东西有独立的一个项目来负责，我觉得思路没问题，更多问题是列表内的东西准确性和完整性，这个是需要人去维护的，通过机器判定我觉得不现实。

1. 请问 SSR plus+ 的项目地址是哪一个？搜索了全名，ssr ssr plus等好像都没搜到一个看起来比较官方的项目。
2. domain list 和 ip list 天然存在不一致，靠 list 是解决不了这个问题，所以我提出的解决建议是不需要 list 的。

Ozzy-hzy, 2021-07-15:
> 1. 请问 SSR plus+ 的项目地址是哪一个？搜索了全名，ssr ssr plus等好像都没搜到一个看起来比较官方的项目。
> 2. domain list 和 ip list 天然存在不一致，靠 list 是解决不了这个问题，所以我提出的解决建议是不需要 list 的。

https://github.com/fw876/helloworld/issues 在这

lynncui00, 2021-07-15:
@Ozzy-hzy  感谢

closed 2021-07-15 by lynncui00