coolsnowwolf/lede #7398: Turbo ACC 导致macos 上的 Parallels Client  远程桌面应用无法使用

suizhenmin, 2021-07-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [是 ] 没有类似的issue

## 2. 详细叙述
macbook 上使用 Parallels Client 远程桌面 win10 主机，使用旧版Turbo ACC 时候正常，更新到支持 ipv6 的版本后  Parallels Client 远程连接必须在设置中降低显示的设置，导致远程链接非常卡顿。原先是有2个版本可选，现在已经没有旧版了，只有L大的魔改版本了。此问题在局域网内不会出现，只有在外网远程连接时才会出现。

[image]

[image]

A：

### (2) 路由器型号和固件版本

[image]

A：

### (3) 详细日志

A：

suizhenmin, 2021-07-16:
原来是2个合并了，可以选择

Mitsuhaxy, 2021-07-16:
qca-nss-ecm

DanielChen008, 2021-07-28:
难道我升级固件后，群晖自带的 DDNS 服务（xxx.myds.me:5000）从外面访问不到了也是这个问题？… 回家试试去

closed 2024-11-19 by coolsnowwolf