coolsnowwolf/lede #7401: 升级5.10.46后8125B连接速度由2.5G变成了1G

BigDog66, 2021-07-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： 5.4.109的时候 路由的网卡是X550 电脑的网卡是8125B  X550和8125B连接后 自适应2.5G 今天升级了5.10.46 电脑上8125B的速率下降到了1GB   X540与X540对接没问题 还是10GB

### (2) 路由器型号和固件版本

A：X86 5.10.46

### (3) 详细日志

A：[    0.000000] Linux version 5.10.46 (lede@ledecode) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r3471-604ceaa75) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Mon Jul 12 16:39:11 2021
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=a9e5513e-96e7-a29c-e74d-3472168eaf02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000ef87fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000ef88000-0x000000000ef8efff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000ef8f000-0x000000000fee7fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fee8000-0x000000000ff07fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000ff08000-0x000000000ff57fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000ff58000-0x000000000ff73fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000ff74000-0x000000000ff77fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000ff78000-0x000000007fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.31 by VMware, Inc.
[    0.000000] efi: SMBIOS=0xff75000 ACPI 2.0=0xff69000 MEMATTR=0xfd90218 
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.17369862.B64.2012240522 12/24/2020
[    0.000000] vmware: hypercall mode: 0x02
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 3000.000 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000000] vmware: using clock offset of 5217196611 ns
[    0.000010] tsc: Detected 3000.000 MHz processor
[    0.001553] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001554] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001557] last_pfn = 0x80000 max_arch_pfn = 0x400000000
[    0.001572] MTRR default type: uncachable
[    0.001572] MTRR fixed ranges enabled:
[    0.001573]   00000-9FFFF write-back
[    0.001573]   A0000-FFFFF uncachable
[    0.001574] MTRR variable ranges enabled:
[    0.001575]   0 base 000000000000 mask 1FFF80000000 write-back
[    0.001575]   1 disabled
[    0.001575]   2 disabled
[    0.001576]   3 disabled
[    0.001576]   4 disabled
[    0.001576]   5 disabled
[    0.001576]   6 disabled
[    0.001577]   7 disabled
[    0.001589] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.006187] Using GB pages for direct mapping
[    0.006391] Secure boot disabled
[... 1136 lines cut ...]
[   20.067253] br-lan: port 2(eth2) entered blocking state
[   20.067649] br-lan: port 2(eth2) entered forwarding state
[   20.068075] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   20.228643] igb 0000:03:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   20.232224] br-lan: port 1(eth0) entered blocking state
[   20.232976] br-lan: port 1(eth0) entered forwarding state
[   21.765495] igb 0000:0d:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   21.771178] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   21.971241] pppoe-wan1: renamed from ppp0
[   23.763616] ixgbe 0000:17:00.1 eth3: NIC Link is Up 10 Gbps, Flow Control: RX/TX
[   23.770027] br-lan: port 3(eth3) entered blocking state
[   23.770416] br-lan: port 3(eth3) entered forwarding state
[   36.982456] squashfs: Unknown parameter 'discard'
[   36.985359] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
[   36.986036] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[   36.986790] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x80a345bd)
[   36.987458] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   36.988586] exFAT-fs (sda128): invalid boot record signature
[   36.989230] exFAT-fs (sda128): failed to read boot sector
[   36.989870] exFAT-fs (sda128): failed to recognize exfat type
[   92.710062] ixgbe 0000:17:00.0 eth2: NIC Link is Down
[   92.720326] br-lan: port 2(eth2) entered disabled state
[   98.572520] ixgbe 0000:17:00.0 eth2: NIC Link is Up 1 Gbps, Flow Control: RX/TX
[   98.578517] br-lan: port 2(eth2) entered blocking state
[   98.578991] br-lan: port 2(eth2) entered forwarding state

suizhenmin, 2021-07-16:
没有这样的情况，我的也是8125b的软路由，一直正常

BigDog66, 2021-07-16:
> 没有这样的情况，我的8125b的也是软路由，一直正常

你没仔细看 我的路由是X550 电脑网卡才是8125B

117104010497, 2022-04-12:
我有遇到
Ubuntu 18.04 kernel 4.15 X550
Ubuntu 20.04 kernel 5.13 8125BG
直连后为1G
请问您有解决吗？

closed 2023-12-22 by aiamadeus