coolsnowwolf/lede #7403: 7月9日之后的源码编译的AX6固件会突发数次重启

Kaze1027, 2021-07-16:
反馈bug

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)

* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用2021年7月9日之后的源码编译出来的ax6固件会在运行大约30分钟左右突然数次重启，但是使用7月9日未添加nss驱动的固件便不会出现该情况。

### (2) 路由器型号和固件版本

A：路由器型号：Redmi AX6，固件版本：OpenWrt R21.7.15 / LuCI Master (git-21.188.53404-317d4ad)

### (3) 详细日志

A：详细日志内容如下：

- kernel.log：

```
2021-07-16T19:00:28+00:00 OpenWrt kernel: [   19.745634] br-lan: port 1(eth1) entered blocking state
2021-07-16T19:00:28+00:00 OpenWrt kernel: [   19.745672] br-lan: port 1(eth1) entered forwarding state
2021-07-16T19:00:28+00:00 OpenWrt kernel: [   19.749958] br-lan: port 2(eth2) entered blocking state
2021-07-16T19:00:28+00:00 OpenWrt kernel: [   19.755303] br-lan: port 2(eth2) entered forwarding state
2021-07-16T19:00:28+00:00 OpenWrt kernel: [   19.760586] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
2021-07-16T19:00:28+00:00 OpenWrt kernel: [   19.766231] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.504535] wlan1: Failed to create a NSS virtual interface
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.594964] wlan0: Failed to create a NSS virtual interface
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.596580] br-lan: port 4(wlan1) entered blocking state
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.599410] br-lan: port 4(wlan1) entered disabled state
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.605280] device wlan1 entered promiscuous mode
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.610355] br-lan: port 4(wlan1) entered blocking state
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.614857] br-lan: port 4(wlan1) entered forwarding state
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.660528] br-lan: port 5(wlan0) entered blocking state
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.660565] br-lan: port 5(wlan0) entered disabled state
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.665511] device wlan0 entered promiscuous mode
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.670497] br-lan: port 5(wlan0) entered blocking state
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.674899] br-lan: port 5(wlan0) entered forwarding state
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.749939] br-lan: port 4(wlan1) entered disabled state
2021-07-16T19:00:29+00:00 OpenWrt kernel: [   20.750599] br-lan: port 5(wlan0) entered disabled state
2021-07-16T19:00:30+00:00 OpenWrt kernel: [   21.368538] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
2021-07-16T19:00:30+00:00 OpenWrt kernel: [   21.368769] br-lan: port 5(wlan0) entered blocking state
2021-07-16T19:00:30+00:00 OpenWrt kernel: [   21.374043] br-lan: port 5(wlan0) entered forwarding state
2021-07-16T19:00:33+00:00 OpenWrt kernel: [   24.437272] pppoe-wan: renamed from ppp0
2021-07-16T19:00:34+00:00 OpenWrt kernel: [   25.727460] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
2021-07-16T19:00:34+00:00 OpenWrt kernel: [   25.727674] br-lan: port 4(wlan1) entered blocking state
2021-07-16T19:00:34+00:00 OpenWrt kernel: [   25.732955] br-lan: port 4(wlan1) entered forwarding state
2021-07-16T19:00:38+00:00 OpenWrt kernel: [   29.866367] device wlan1 left promiscuous mode
2021-07-16T19:00:38+00:00 OpenWrt kernel: [   29.866492] br-lan: port 4(wlan1) entered disabled state
2021-07-16T19:00:49+00:00 OpenWrt kernel: [   40.357624] pppoe-wan: renamed from ppp0
```

- dnsmasq.log:

```
2021-07-16T19:00:30+00:00 OpenWrt dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
2021-07-16T19:00:30+00:00 OpenWrt dnsmasq: Allowing 127.0.0.0/8 responses
2021-07-16T19:00:32+00:00 OpenWrt dnsmasq[3643]: Connected to system UBus
2021-07-16T19:00:34+00:00 OpenWrt dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
2021-07-16T19:00:34+00:00 OpenWrt dnsmasq: Allowing 127.0.0.0/8 responses
2021-07-16T19:00:36+00:00 OpenWrt dnsmasq[4662]: Connected to system UBus
2021-07-16T19:00:38+00:00 OpenWrt dnsmasq[4921]: Connected to system UBus
```

- odhcpd.log:
[... 198 lines cut ...]
2021-07-16T19:00:55+00:00 OpenWrt miniupnpd[8240]: HTTP listening on port 5000
2021-07-16T19:00:55+00:00 OpenWrt miniupnpd[8240]: Listening for NAT-PMP/PCP traffic on port 5351
```

- nlbwmon.log:

```
2021-07-16T19:00:54+00:00 OpenWrt nlbwmon: Reloading nlbwmon due to ifup of lan (br-lan)
2021-07-16T19:01:05+00:00 OpenWrt nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
2021-07-16T19:15:02+00:00 OpenWrt nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
2021-07-16T19:15:14+00:00 OpenWrt nlbwmon: Reloading nlbwmon due to ifup of wan_6 (pppoe-wan)
```

- fwknopd.log:

```
2021-07-16T19:00:34+00:00 OpenWrt fwknopd[4420]: No PCAP_INTF interface specified in fwknopd.conf, fwknopd's default  will be used
2021-07-16T19:00:34+00:00 OpenWrt fwknopd[4420]: Postponing start-up of fwknopd, network  is not up
```

- crond.log:

```
2021-07-16T19:00:38+00:00 OpenWrt crond[4979]: crond (busybox 1.33.1) started, log level 8
```

Kaze1027, 2021-07-17:
[image]

Kaze1027, 2021-07-17:
[image]

Kaze1027, 2021-07-17:
[image]

Kaze1027, 2021-07-17:
[image]

Kaze1027, 2021-07-17:
[image]

qjiang888, 2021-07-18:
我k2p稳定用openwrt几年了，这个最新源码编译的 无缘无故会重启，今天下午上莫名奇妙很卡，只能重启

Kaze1027, 2021-07-19:
我发现设备下面不接任何有线可以有效缓解这个问题

closed 2021-11-30 by AmadeusGhost