coolsnowwolf/lede #7409: 测速组件在python 3.9 下运行失败

HoldOnBro, 2021-07-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：测速组件在python 3.9 下运行失败

### (2) 路由器型号和固件版本

A：所有型号，7月16日最新版本

### (3) 详细日志

A：

```
Retrieving speedtest.net configuration...
Traceback (most recent call last):
  File "/usr/bin/speedtest", line 33, in <module>
    sys.exit(load_entry_point('speedtest-cli==2.1.2', 'console_scripts', 'speedtest')())
  File "/usr/lib/python3.9/site-packages/speedtest.py", line 1986, in main
  File "/usr/lib/python3.9/site-packages/speedtest.py", line 1872, in shell
  File "/usr/lib/python3.9/site-packages/speedtest.py", line 1091, in __init__
  File "/usr/lib/python3.9/site-packages/speedtest.py", line 1173, in get_config
ValueError: invalid literal for int() with base 10: ''
```

hxlls, 2021-07-17:
?这。。不应该去speedtest.py的项目那里提么？

shinegod, 2021-07-17:
pip uninstall speedtest-cli
pip install speedtest-sli
重新安装一下不就好了。

shinegod, 2021-07-17:
pip install speedtest-cli 打错字母了

shinegod, 2021-07-17:
curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
python get-pip.py
没有pip
自己这样安装pip

HoldOnBro, 2021-07-17:
> ?这。。不应该去speedtest.py的项目那里提么？

源项目已经适配了，只是lede里面的版本太老

HoldOnBro, 2021-07-19:
已PR。